ccMSC/ckb #220: CAPS LOCK and NUM LOCK Behave Strangely on Strafe

hyph3n, 2016-01-14:
The Corsair Strafe keyboard has illuminated indicators for num lock and caps lock, however pressing either of the num lock or caps lock keys does not toggle the state of the illuminated indicator. In fact, it causes a temporary "freeze" of the selected light pattern.

Scenario:

I'm typing normally on my keyboard, and I see the "num lock" indicator is lit, and the "caps lock" indicator is not. I currently have a "ripple" lighting effect set up for each key I type.

If I press "caps lock", I would expect the light to ripple out from the "caps lock" key, and for the "caps lock" indicator to become illuminated.

What actually happens is that the "caps lock" key lights up for a few seconds while all other keys remain dark. The "caps lock" indicator does not illuminate. When I type, letters are indeed capitalized, and after a few seconds, the keyboard's "ripple" effect starts responding normally again, but the indicator never lights up to show that "caps lock" is engaged.

The same type of behavior happens if I click on "num lock" (not that letters get capitalized, but that it's clear I've changed the "num lock" state, and yet the keyboard doesn't react as you'd expect).

Fedora 22
linux 4.2.8-200.fc22.x86_6
ckb beta-v0.2.2

Thanks!

ccMSC, 2016-01-15:
There have been a few changes related to this in the v0.2.3 update that I just released. Try installing the new version and see if it fixes the problem.

hyph3n, 2016-01-15:
I upgraded to version 0.2.3 this morning, and it appears the lighting animation on the keyboard no longer works at all. When I have CKB's GUI open and I press a key, I see the ripple effect previewed in the application, but nothing happens on the keys. I tried clicking "Save to hardware", but this just results in a momentary flash of the lights on the keys.

The "caps lock" and "num lock" indicators still do not light when I press either of those keys.

I upgraded by cloning the repo and executing ./quickinstall as root. Following the installation, I rebooted the machine.

I'll keep playing with it to see what I can figure out.

Thanks.

hyph3n, 2016-01-15:
UPDATE: I manually built ckb rather than using the quickinstall script, and I now have the keyboard animation working properly. Original issue persists as reported. Let me know if there's anything I can do to help.

Thanks.

ccMSC, 2016-01-16:
In the Performance tab there's an option to use an RGB indicator (shown on the caps lock key itself) as opposed to the regular indicator LED. If you switch to that, does it solve the problem?

Otherwise, could you provide a log from the daemon? From Terminal, you can start it with:
```
sudo systemctl stop ckb-daemon
sudo ckb-daemon
```
Then press the caps lock key and see if there's any output.

hyph3n, 2016-01-16:
I enabled "RGB Indicator" for both lock keys in the GUI, and set the "On" color value to #FF0000, since this Strafe only has red leds.

The light animation no longer freezes when these keys are pressed, but what does happen is that the key itself remains lit rather than the indicator light at the top right of the keyboard. ie. When caps lock is off, the caps lock key is dark and never lights, even when the ripple animation hits it. When caps lock is on, the caps lock key is lit, and remains lit until caps lock is turned off. Same behavior for num lock. All the while, the status indicator LEDs stay in the same state they were in at boot (num lock is lit, and caps lock is dark).

I set both lock lights back to "Normal" in the GUI, and observed the originally reported behavior. I then ran the daemon in a terminal but got no unusual output when pressing keys:

    ckb: Corsair RGB driver beta-v0.2.3
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair STRAFE Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usbsend (via firmware.c:15): Connection timed out
[E] os_usbsend (via firmware.c:15): Connection timed out
[E] os_usbsend (via firmware.c:15): Connection timed out
[E] os_usbsend (via firmware.c:15): Connection timed out
[E] os_usbsend (via firmware.c:15): Connection timed out
[E] _usbrecv (via firmware.c:15): Too many send/recv failures. Dropping.
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1

I didn't start typing/testing the caps lock and num lock keys until after the "Setup finished" line, so while the errors and warning might be relevant, they weren't caused by my keypresses.

Anyway, this issue isn't a deal breaker for me by any means. If you want to continue to pursue it, I'm more than happy to help, but I'm sure you have higher priorities than hunting down why indicator lights aren't illuminating. Thanks for the effort you put into creating ckb.

ccMSC, 2016-01-17:
Seems I forgot to add logging code in case the indicator LEDs fail to send. Seems odd that it would happen at all, considering it's a standard HID message. I've updated the testing repo so it should now produce some output when you press the caps lock key.

One other thing: could you give me the output of `lsusb -vd 1b1c:`? It prints a lot of information but it might help me figure out what's going on here.

hyph3n, 2016-01-17:
Thanks for the reply. I built and ran the testing branch, and I get the following:

>    ckb: Corsair RGB driver beta-v0.2.3+t01
>[I] Root controller ready at /dev/input/ckb0
>[I] Connecting Corsair STRAFE Gaming Keyboard at /dev/input/ckb1
>[I] Starting input thread for /dev/input/ckb1
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_usbsend (via firmware.c:15): Connection timed out
>[E] os_usbsend (via firmware.c:15): Connection timed out
>[E] os_usbsend (via firmware.c:15): Connection timed out
>[E] os_usbsend (via firmware.c:15): Connection timed out
>[E] os_usbsend (via firmware.c:15): Connection timed out
>[E] _usbrecv (via firmware.c:15): Too many send/recv failures. Dropping.
>[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
>[I] Setup finished for /dev/input/ckb1
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_sendindicators (usb_linux.c:84): Connection timed out
>[E] os_sendindicators (usb_linux.c:84): Connection timed out

The first few "os_sendindicators" errors happened after starting the daemon without any interaction on my part, but now I see a new message appear consistently when I press either of the keys in question.

I've attached a file containing the output of lsusb here: [lsusb-output.txt](https://github.com/ccMSC/ckb/files/93417/lsusb-output.txt)

You didn't ask for it, but I also thought I'd pass along all the output of ```dmesg | grep 8-2``` (since that's the USB bus the keyboard is connected to). Other than redacting my serial number, everything else is intact since my last reboot. Starting at [76827.164177] you can see where I killed the old daemon and executed the new testing build: [dmesg-output.txt](https://github.com/ccMSC/ckb/files/93425/dmesg-output.txt)

Thanks.

ccMSC, 2016-01-20:
Thanks. I'm not noticing any obvious indications as to what's going wrong, but I'll let you know as soon as I figure something out.

hyph3n, 2016-06-15:
Hi, is this fixed, or is there some other update? Simply closing the bug doesn't really give me a whole lot of information. Thanks!

ksingh7, 2016-06-29:
Hey guys i have exact same problem , 15 minutes back i have received my Corsair Strafe and i am using OSx. I too have Caps Lock and Num Lock LED's not glowing.  @ccMSC any hits , i am sorry i am not sure if your CKB thing is only Linux specific. But looking forward for any hints.

ccMSC, 2016-06-30:
Sorry, no it wasn't fixed. I was doing a mass-close of issues that never went anywhere but I shouldn't have closed this one. I'll re-open it.

That said, the reason I never updated is because I really have no idea what's causing this. Do either of you have Windows on your machines and if so, could you provide a USB capture from both Windows and Linux so I can compare them?

hyph3n, 2016-07-05:
I do have a Windows partition, though I very rarely use it. I'll try to make a point to boot to it soon. Do you just want a Wireshark USB capture? Should I hit a sequence of reference keys before hitting the problem keys? Just want to make sure I get you something useful.

Thanks!

zicodxx, 2016-07-24:
I ran into a similar issue with the Strafe RGB. But however for me, only the Lock LEDs itself "froze" (i.e. didn't toggle on/off immediately or only after a new button press). Some Windows users seem to report similar issues and fixing them by using USB 3 connectors (which do not work for me at all - my mainboard is picky).

Still I found a workaround by changing the FPS in the GUI from 30 to 20 or even down to 15 (which I still think looks smooth enough). I am wondering if this may solve the issue for you guys. I know it's not exactly the same problem but maybe it still helps?

chjohnst, 2017-03-21:
If I hit caps lock on the latest drivers, it almost freezes my system.  Kinda frustrating since the only cure is to reboot it.