ccMSC/ckb #221: Strafe RGB: Wrong key on Key binding screen (Right Cmd)

shazron, 2016-01-18:
Version:
ckb beta-v0.2.3

Keyboard:
Corsair Strafe RGB w/ latest firmware as of this date

Description:
The key `FN` on the front row (see screenshot attached, its the key to the left of the `Menu` key) is represented on the `Binding` tab as `Cmd`, and bound to `Right Cmd` by default. It is not actually bound to `Right Cmd`, a Cmd-Spacebar will test this (it should show the Spotlight searchbar in the middle of the screen, but does not)

Expected:
The key label should show up as `FN` and by default is bound to the `FN` keycode

Actual:
The key label is `Cmd` (clover key) and by default is bound to the `Right Cmd` keycode in the UI (but not actually bound to that keycode)

[image]

SicVolo, 2016-01-18:
Fixed in pull #223. The UI is correct, but Fn keys does not seem to do much even after a proper re-bind.

shazron, 2016-03-03:
Fixed in 0.24

closed 2016-03-03 by shazron