ccMSC/ckb #222: Strafe RGB: swapping Alt and Cmd does not swap keys on the right side of the spacebar

shazron, 2016-01-18:
Related Issue:
#221 Strafe RGB: Wrong key on Key binding screen (Right Cmd)

Version:
ckb beta-v0.2.3

Keyboard:
Corsair Strafe RGB w/ latest firmware as of this date

Description:
In the `Settings` tab, when you swap the bindings for the `Windows` (changed from Command to Option) and `Alt` (changed from Option to Command) keys globally, the keys on the right side of the spacebar are swapped in the `Bindings` tab UI but only the `Alt` key is swapped programmatically, not the` FN ` key. The two keys on the right side of the spacebar should not be swapped (see #221, Right-Cmd is actually FN)

[image]

[image]

Expected:
The bindings and UI for the two keys on the right side of the spacebar are not swapped and stay as is (because of #221)

Actual:
The bindings for the two keys on the right side of the spacebar are swapped as indicated in the `Bindings` tab UI, but only `Alt` is swapped for `Cmd` (FN is not swapped)

SicVolo, 2016-01-18:
Please try the fix from pull #223, after it makes into CKB's testing, or directly from my repo - https://github.com/SicVolo/ckb/tree/testing

shazron, 2016-03-03:
Fixed in 0.24

closed 2016-03-03 by shazron