ccMSC/ckb #225: Strafe RGB: Keyboard not detected after swapping back to OS X from a USB switch

shazron, 2016-01-21:
Version:
ckb beta-v0.2.3

Keyboard:
Corsair Strafe RGB w/ latest firmware as of this date

Description:
I have a Windows and OS X system that I use a [USB switch](http://www.amazon.com/gp/product/B00JX3Q28Y)[1] for, to switch my peripherals (keyboard, mouse, iPhone dock, etc). The Windows system has the Corsair CUE software installed, and has no problem with me using the USB switch to switch back and forth. After switching back to OS X from the Windows system, ckb cannot detect the Strafe RGB keyboard until I force-quit the ckb-daemon process (then somehow it resets itself).

Expected:
The keyboard is detected after a USB switch swap.

Actual:
The keyboard is not detected after a USB switch swap.

Workaround:
Force quit the ckb-daemon process in Activity Monitor.

[1] http://www.amazon.com/gp/product/B00JX3Q28Y

Sparta1534, 2016-01-21:
I am having a similar issue with my mouse, I am also on an OSX system.  How did you force quit the Daemon?

SicVolo, 2016-01-21:
I don't think this is fixable - CUE ties KB to it, CKB driver need to re-init to cut the ties. 
There is a much better solution though than using a physical switch - Synergy (http://synergy-project.org/). Run your keyboard and mouse permanently attached to OSX and switch to Windows in software. Works perfectly with CKB.

shazron, 2016-01-21:
@Sparta1534 launch the app "Activity Monitor". View -> All Processes. In the search field, type "ckb-daemon". Select it, then press the "[X]" button on the toolbar, then select Force Quit.

@SicVolo thanks for the pointer to Synergy, I'll check it out for sure.

ccMSC, 2016-01-21:
@SicVolo is correct. Someone on Linux had the same problem: #172 

Basically, the keyboard wasn't designed to be KVM-compatible and unless Corsair takes steps to implement compatibility, it probably never will be.

shazron, 2016-01-21:
Thanks everyone, I'll close this. 
I usually have a terminal window open and will just do a `sudo killall -9 ckb-daemon` whenever I need to.

closed 2016-01-21 by shazron