ccMSC/ckb #226: Mouse 4 & 5

Hotsauze, 2016-01-21:
Hi I'm running OSX El Capitan and just recently got the M65 mouse. Downloaded beta v0.2.3. Everything works OK except the Back (4) and Forward (5) doesn't seem to work when browsing Safari or Chrome. It is binded correctly but it just doesn't do anything. Is anyone else having the same issues? Would really appreciate a solution. 

Thank you.

ccMSC, 2016-01-21:
Unfortunately, the majority of OSX apps don't recognize the back/forward buttons at all. I don't currently know of a workaround for this.

Hotsauze, 2016-01-24:
If you allow for multiple key bindings i.e Command + [ or Command + ] then the forward n back in a webpage will work? Right now I've bound [ to the mouse 4 n have to hold the command key n click to go Back. Thanks

closed 2016-06-13 by ccMSC

jvanhees, 2016-08-19:
Just to commend on this, I solved with an AppleScript:

Under 'Bindings', select 'Program' and insert the following line in the key press or key release input field:

`osascript -e 'tell application "System Events" to keystroke "[" using {command down}'`

`osascript -e 'tell application "System Events" to keystroke "]" using {command down}'`

This will execute the applescript that will tell the system events application that "]" or "[" has been pressed with command down. This propogates to all other programs like Chrome, Terminal or Finder.