ccMSC/ckb #232: El Capitan issues

smittix, 2016-01-28:
Hi I've been using this with no problem for months, after updating to the latest release it seems to randomly lose animation settings for both K70 and M65. 

Has anyone else reported this?

ccMSC, 2016-01-29:
Not that I'm aware of. If you close the app from the Settings screen and then restart it, that should force the settings to sync.

smittix, 2016-01-29:
I tried that but to avail. I had to recreate my profiles. Even the standard ones are gone?

DevDanSTL, 2016-01-30:
Same here, after updating to El Capitan I can only see the settings tab and can't get to animations. Closing and restarting the app, rebooting and force killing it haven't fixed it. Think someone said that the program doesn't appear to detect the keyboard anymore and that's in line with what I'm seeing with my K70.

ccMSC, 2016-02-01:
@DevDanSTL I think you're experiencing a different issue. If you can only see the settings tab, the keyboard isn't being detected by the driver. Have a look at #233.

@smittix If it happens again, could you try deleting the preferences file at `~/Library/Preferences/com.ckb.ckb.plist`? It *should* reload the demo profile if it's not able to read any preferences, so there might be something corrupted in there.

closed 2016-06-13 by ccMSC