ccMSC/ckb #233: ckb does not detect keyboard after restart of OS X

Lockjututz, 2016-01-29:
I've used ckb to swap the alt and option keys on my K70 RGB but when I restart the computer (Mid 2014 15" retina MBP with OS X 10.11.3) the keys are reverted to the old layout. The lighting scheme is the same, however. Furthermore, when I open ckb (using "restore" from the autostarted app) it only shows the settings tab. I've verified that the keyboard is connected in System Report (as well as writing this on it) but it seems that ckb does not detect the keyboard.

This issue has occured on all restarts I've done (two as of now). I solved it last time by executing a terminal command that reloaded the driver (I think) but I don't remember the actual command.

Is this a known issue? I can help with debugging if needed.

ccMSC, 2016-02-01:
Please post a driver log following these instructions: https://github.com/ccMSC/ckb/issues/106#issuecomment-131287945

2-alex-2, 2016-02-12:
I'm now trying to find the logs. but doing the manual input has given me my keyboard recognition on boot up but ckb still not recognising where will i find the log file.

shanece, 2016-02-16:
I've i was checking to see if there is a resolve for the same issue I am facing. I reference this one (#233) and #240 but they seem to link to each other but not hint to any resolve or trouble shoot. 

I have a STRAFE RGB mech keyboard and I downloaded CKB. When first install, I saw that the perphial was connected and I can customize as need. But with in a few minutes of use, the keyboard was no longer being acknowledged by the driver and I tried plugging the keyboard in and out with no further success. When the system restarts, I can use my keyboard for BIOS and bootloader menu but once my hackintosh loads to the mac os x I lose keyboard connectivity and have to plug in another keyboard to work. To get the mechanical keyboard to work again I have to go to unplug/plug in the keyboard then go to system preference to reconfigure the keyboard. Do you now a work around on this, to fix the issue?

2-alex-2, 2016-02-17:
I modified the file manually and keyboard gets recognised by the is and works but ckb won't work.

2-alex-2, 2016-02-17:
Modify the lunch demon with this and it might work from boot but ckb might not recognise it. 

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>com.ckb.daemon</string>
    <key>ProgramArguments</key>
    <array>
        <string>/Applications/ckb.app/Contents/Resources/ckb-daemon</string>
    </array>
    <key>KeepAlive</key>
    <true/>
    <key>ProcessType</key>
    <string>Interactive</string>
    <key>LegacyTimers</key>
    <true/>
    <key>StandardOutPath</key>
    <string>/var/log/ckb.log</string>
    <key>StandardErrorPath</key>
    <string>/var/log/ckb.err.log</string>
</dict>
</plist>

2-alex-2, 2016-02-17:
Copy it from #106 it didn't paste right

2-alex-2, 2016-02-17:
Mac-Pro:~ Alex$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.4
[I] Root controller ready at /var/run/ckb0
[I] Connecting  at /var/run/ckb1
[I] Starting input thread for /var/run/ckb1
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
sudo /Applications/ckb.app/Contents/Resources/ckb-dae[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] _usbrecv (via firmware.c:15): Too many send/recv failures. Dropping.
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /var/run/ckb1

i also had this pop up one time today whilst looking at trying to get it to work

shanece, 2016-02-18:
I tried plugging the keyboard into my US3.0 port and it worked to the point of being recognized at boot and system startup. 
The only issue now is getting ckb to recognize the keyboard. Any suggestion for that?

ccMSC, 2016-02-18:
@2-alex-2 the log files are named `ckb.log` and `ckb.err.log` and they can be found in `/var/log`. You can get there in Finder using Cmd+Shift+G and then typing `/var/log` into the prompt. You may also need to stop the driver using a Terminal `sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist` in order for the files to appear.

The log you posted looks normal, seems like it had no issues setting it up that time.

2-alex-2, 2016-02-19:
The command unload couldn't find specified service. 

And then /var/log folder can't be found.

2-alex-2, 2016-02-21:
still no luck I'm afraid can't find the log files. but if i enter

sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist

in terminal once its started it work again and recognises the keyboard just something there in the program that doesn't work on restart.

shanece, 2016-02-21:
I tried the same. I uninstalled the program and reinstalled and it worked on launch, but when I restarted the pc up again it was unrecognizable yet again.

ccMSC, 2016-02-24:
I may have found the cause of this. I'm still in the middle of troubleshooting it but I'll have a new version up soon.

rhardman, 2016-02-24:
I'm having similar issues, this is what I get from running interactively.

$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon                       

    ckb: Corsair RGB driver beta-v0.2.4
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard  at /var/run/ckb1

<<< Removed line feeds >>>

[E] iterate_devices (usb_mac.c:413): Failed to create device plugin: e00002be

<<< Removed line feeds >>> (Toggled switch on keyboard)

[I] Disconnecting /var/run/ckb1

2-alex-2, 2016-03-02:
any news on the update

ccMSC, 2016-03-03:
On hold for the moment as I have to rewrite the USB code in order to deal with the new Scimitar firmware. I'll post a test version here as soon as it's available.

2-alex-2, 2016-03-03:
yea that not a problem take as much time as you need wasn't sure if you had a time scale in mind thats all. i use to code but its change a lot these days and wouldn't know where to start now plus its mac only just started on mac with a hackintosh.

ccMSC, 2016-03-08:
I was hoping to get it done over the weekend but that didn't happen. I'll have more time to work on it this week though, so hopefully within the next 5-6 days.

ccMSC, 2016-03-14:
FWIW, I've pushed the updates to the `fw2` branch. Here's a binary link: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.4%2Bf02.pkg

I'm not sure I got the fix I wanted. It's still incomplete at the moment. Try it out and let me know if you have any luck, though.

2-alex-2, 2016-03-14:
its so close cob recognise the keyboard but the keys done type anything. but if you change the refresh rate on the back of the keyboard you can then type. also noticed some key don't return to the colour they are set to. like my wasd is green the rest are red but i have a rainbow effect when the keys are pressed and some time the wasd don't always return to green.

ccMSC, 2016-03-15:
That's exactly what I suspected, thanks for confirming. I'll have another test build available in a few days.

2-alex-2, 2016-03-15:
But it's a lot less hassle to flick the switch on the back then got bh into terminal all the time. Wish I knew where to start with helping lol coding on a Mac I don't have a clue.

ccMSC, 2016-03-22:
new version is ready!

https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.4%2Bf03.pkg

Should be fixed properly now. Let me know if it works.

2-alex-2, 2016-03-22:
no once the pc starts up and boots into os light all come on but no animations. open ckb and its recognised the keyboard but still no key inputs until you flick the switch on the back of the keyboard. once thats done it all work as it should.

ccMSC, 2016-03-23:
Try replacing the launchdaemon with this:

`/Library/LaunchDaemons/com.ckb.daemon.plist`

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>com.ckb.daemon</string>
    <key>ProgramArguments</key>
    <array>
        <string>/Applications/ckb.app/Contents/Resources/ckb-daemon</string>
    </array>
    <key>KeepAlive</key>
    <true/>
    <key>ProcessType</key>
    <string>Interactive</string>
    <key>LegacyTimers</key>
    <true/>
    <key>StandardOutPath</key>
    <string>/Applications/ckb.app/Contents/ckb-daemon.log</string>
</dict>
</plist>
```

To find the log file, go to your Applications folder, right-click on `ckb`, and select `Show Package Contents`. Then click on the `Contents` folder and you should see a file named `ckb-daemon.log`.

2-alex-2, 2016-03-23:
that has now back to normal with no animations even if you flick the switch on the back and there is no log file that i can see

ccMSC, 2016-03-26:
Sounds like it stopped running completely?

Here's a version with logging enabled by default: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.4%2Bf03%2Blogging.pkg

Install that over the old one and it should work. If there's still no log, try running these commands in a Terminal:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo rm -f /Library/LaunchDaemons/com.ckb.daemon.plist
sudo rm -Rf /Applications/ckb.app
```

then reinstall it again.

2-alex-2, 2016-03-27:
ckb: Corsair RGB driver beta-v0.2.4+f03
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifhid[3]
[I] Adding ckb1.ifhid[2]
[I] Adding ckb1.ifhid[1]
[I] Adding ckb1.ifhid[0]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifhid[1]
[I] Reading ckb1.ifhid[2]
[I] Setup finished for /var/run/ckb1

[I] Caught signal 15
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb: Corsair RGB driver beta-v0.2.4+f03
[I] Root controller ready at /var/run/ckb0
[I] Connecting  at /var/run/ckb1
[E] iterate_devices_usb (usb_mac.c:577): USBDeviceOpen failed: e00002c5
[I] Connecting Corsair K65 RGB Gaming Keyboard at /var/run/ckb1
[I] Adding ckb1.ifhid[0]
[I] Adding ckb1.ifhid[2]
[I] Adding ckb1.ifhid[1]

2-alex-2, 2016-03-27:
also now get this on f2 which worked the best before.
Last login: Sun Mar 27 21:36:53 on console
Mac-Pro:~ Alex$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon 
Password:
    ckb: Corsair RGB driver beta-v0.2.4+f02
[F] ckb-daemon is already running (PID 63). Try `killall ckb-daemon`.
[F] (If you're certain the process is dead, delete /var/run/ckb0/pid and try again)
Mac-Pro:~ Alex$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon 
    ckb: Corsair RGB driver beta-v0.2.4+f02
[I] Root controller ready at /var/run/ckb0
[E] iterate_devices_usb (usb_mac.c:465): USBDeviceOpen failed: e00002c5
[E] iterate_devices_hid (usb_mac.c:636): Failed to seize device: e00002c5
[E] iterate_devices_hid (usb_mac.c:636): Failed to seize device: e00002c5

ccMSC, 2016-03-31:
Thanks! Made some more adjustments, new binary available here: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.4%2Bf04%2Blogging.pkg

Logging should still be enabled, same location as before.

2-alex-2, 2016-04-01:
that much better will run it for a few days see if anything pops up at all. still some times keys don't return to colour they should do. like my wasd is green and rest is red but when a key is pressed that key does a rainbow colour but doesn't always return back to green on my wasd. other than that seem ok here is the log file anyway but its working with out doing anything in terminal or flicking a switch. 

[I] Caught signal 15
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb: Corsair RGB driver beta-v0.2.4+f04
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifusb[1]
[I] Adding ckb1.ifusb[2]
[I] Adding ckb1.ifusb[3]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002c7
[W] iterate_devices_usb (usb_mac.c:626): Unable to seize USB handle, continuing anyway...
[W] add_usb (usb_mac.c:514): Tried to set up handle for device ckb1, but it was already set up. Skipping...
[I] Adding ckb1.ifhid[0]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifusb[1]
[I] Reading ckb1.ifusb[2]
[W] iterate_devices_usb (usb_mac.c:626): Unable to seize USB handle, continuing anyway...
[E] os_usbrecv (via profile_keyboard.c:35): Got return value 0xe0004051
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Connecting  at /var/run/ckb2
[I] Setup finished for /var/run/ckb1

ccMSC, 2016-04-04:
Good to hear. I'm going to freeze this for now as I'd like to roll out the firmware update for Scimitar users, but I'll continue to work on it in future versions. Thanks for your help.

nbranch1311, 2016-04-12:
@ccMSC I am on 10.11.4 and I am having this same issue. My keyboard is not being detected and I followed all the troubleshooting steps. It does not work on restart and that is including in BIOS mode.

Was there a fix or is that in the works. Thank you for all your hard work on this.

2-alex-2, 2016-04-14:
What version are you running as a fix for me was implemented into the latest update

nbranch1311, 2016-04-14:
@2-alex-2 ckb beta v0.2.5 is this correct

nbranch1311, 2016-04-14:
@ccMSC and @2-alex-2 

I used the latest update. uninstalled the old one and did the troubleshoot and still nothing. I am hoping this is something I am doing. just says No devices connected

2-alex-2, 2016-04-15:
i have got issues now on the 4 reboot i have nothing no keyboard or light effects and no connected again. wired as first three reboots were fine

nbranch1311, 2016-04-15:
@2-alex-2 ... well this isnt good lol. Dang it I was really hoping I could get this damn thig to work to.

nbranch1311, 2016-04-15:
@ccMSC and @2-alex-2 I think part my problem may be that missing folder /Library/LaunchDaemons/

2-alex-2, 2016-04-15:
Try downloading the f3 version further up it should work ok you may have to flick the rate switch at the back to get it working but other than that was working fine on f3.

nbranch1311, 2016-04-15:
@2-alex-2 can you post the f3. I dont know what that is

2-alex-2, 2016-04-16:
If you look back up this topic there are links to other downloads one one of them has in the link bf03 and logging

EvN-SKO, 2016-04-17:
@2-alex-2 same issue here

is there a possibility that the firmware is outdated and needs an update?

BradAbard, 2016-04-17:
I am unable to get my k70 RGB to work with my iMac running 10.11.4.  The ckb page only has a settings tab that reports "No Devices Connected". 

Tried version 0.2.5 and am now trying 0.2.4+f03

Tried plugging/unplugging in different USB ports.  (only have USB2) 

When the daemon loads, the log says. 
```
> iMac:ckb.app INGLIS$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon 
>     ckb: Corsair RGB driver beta-v0.2.4+f03
> [I] Root controller ready at /var/run/ckb0
> [I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb1
> [I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb2
> [I] Adding ckb2.ifhid[1]
> [I] Adding ckb2.ifhid[3]
> [I] Adding ckb2.ifhid[0]
> [I] Adding ckb2.ifhid[2] 
```
There are no directories at /var/run/ckb1 or ckb2?   The front end reports that no devices are connected.  

Help?

hanskhe, 2016-04-18:
I'm having the same problem on my MacBook Air 2011 running 10.11.4. No issues on my MacBook Pro 2011, running 10.10.5.

I can type with the keyboard, but CKB reports "No devices connected"

ccMSC, 2016-04-18:
@BradAbard I assume there's no mention of device setup finishing?

here's a daemon with the 2.5.0 fixes that has logging enabled by default: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.5-logging.pkg

You should be able to find the log files inside `ckb.app` as described above. Please paste them here after rebooting.

nbranch1311, 2016-04-18:
So I assume no fix

BradAbard, 2016-04-19:
@ccMSC There was no mention of setup finishing. 

I have installed 2.5.0 with logging. 

```
    ckb: Corsair RGB driver beta-v0.2.4+f03
[I] Root controller ready at /var/run/ckb0
[I] Connecting  at /var/run/ckb1
[I] Adding ckb1.ifusb[2]
[I] Adding ckb1.ifusb[3]
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb2
[I] Adding ckb2.ifhid[1]
[I] Adding ckb2.ifhid[0]
[I] Disconnecting /var/run/ckb2

[I] Caught signal 15
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb1
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb2
[I] Adding ckb2.ifhid[1]
[I] Adding ckb2.ifhid[3]
[I] Adding ckb2.ifhid[0]
[I] Adding ckb2.ifhid[2]

[I] Caught signal 15
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifusb[1]
[I] Adding ckb1.ifusb[2]
[I] Adding ckb1.ifusb[3]
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb2
[I] Adding ckb2.ifhid[0]
[W] iterate_devices_hid (usb_mac.c:748): Failed to open device: e00002c5
[I] Disconnecting /var/run/ckb2
 
```
After reboot the keyboard would not work. I had to unplug and replug it in again. 

Any help you can give would be appreciated.

ccMSC, 2016-04-21:
Definitely a bug. Based on those status messages it should be connecting the device at that point, but instead it seems to think it has two different devices. I'll try to have an update on this in a few days.

2-alex-2, 2016-04-21:
ok so this is my log file looks ok and ckb picks up the keyboard. 

[W] Ignoring signal 15 (already shutting down)
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifusb[2]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002c7
[I] Adding ckb1.ifhid[1]
[I] Adding ckb1.ifhid[0]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifhid[1]
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1
[W] iterate_devices_usb (usb_mac.c:626): Unable to seize USB handle, continuing anyway...
[W] add_usb (usb_mac.c:514): Tried to set up handle for device ckb1, but it was already set up. Skipping...
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Connecting Corsair K65 RGB Gaming Keyboard at /var/run/ckb1
[I] Adding ckb1.ifhid[0]
[I] Adding ckb1.ifhid[1]
[I] Adding ckb1.ifusb[2]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifhid[1]
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1
  

but on this the keyboard does not type unless you close ckb

2-alex-2, 2016-04-21:
this was the second boot and same again ckb detects the keyboard but no inputs when ckb is running quit ckb and keyboard works just no animations. 

[W] Ignoring signal 15 (already shutting down)
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifusb[1]
[I] Adding ckb1.ifusb[2]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002c7
[W] iterate_devices_usb (usb_mac.c:626): Unable to seize USB handle, continuing anyway...
[W] add_usb (usb_mac.c:514): Tried to set up handle for device ckb1, but it was already set up. Skipping...
[I] Adding ckb1.ifhid[0]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifusb[1]
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1
[W] iterate_devices_usb (usb_mac.c:626): Unable to seize USB handle, continuing anyway...
[W] add_usb (usb_mac.c:514): Tried to set up handle for device ckb1, but it was already set up. Skipping...

2-alex-2, 2016-04-21:
this is my third boot same as the other boots but different log again. also must add that my keyboard is upto date.

[W] Ignoring signal 15 (already shutting down)
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifusb[2]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002c7
[I] Adding ckb1.ifhid[1]
[I] Adding ckb1.ifhid[0]
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifhid[1]
[W] iterate_devices_usb (usb_mac.c:626): Unable to seize USB handle, continuing anyway...
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb2
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1

2-alex-2, 2016-05-01:
Any new on a fix for the latest issue I have just turns ckb off for now as it types that way.

ccMSC, 2016-05-10:
Really sorry about this, I got side-tracked dealing with the new keyboard firmwares and then I haven't had time to work on the project for a while. I have some ideas about what to do with this, so look for a new test build in the next 1-2 days.

ccMSC, 2016-05-14:
Hi, sorry again, here's the test: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.5%2Bf05%2Blogging.pkg

Logging enabled same as before. I'm not sure whether or not this will solve the issue, but if it doesn't, it should at least tell me where the problem is so that I can get it fixed now.

Stoupik, 2016-05-14:
Hi,
ckb v0.2.5+f05+logging.pkg work good for me now, I have Strafe RGB, last FW 2.04 and OS X 10.11.4.
Thank you very much.

WifiTrash, 2016-05-15:
@ccMSC 
this fix has also worked for me, but since i'm still on FW 1.33, animations glitch out when ckb is open
anyone know where i can find firmware updates and manually download them?
(i have no access to a windows computer near me :()

tatokis, 2016-05-15:
You can find a list of download links directly from corsair in the FIRMWARE file on the root of the repository. 
Despite the version numbers in the file, the links have the latest version

2-alex-2, 2016-05-17:
Seems ok sofar will give it some more reboots and see how it holds up.

2-alex-2, 2016-05-25:
yea still going ok so seems like you have sorted it now. thanks

[1 more reply]

closed 2016-06-13 by ccMSC