ccMSC/ckb #234: CKB Causes long boot time [linux]

kimamb, 2016-01-31:
I was wondering why my PC was starting up slower than it used to, then I checked dmesg, and it says something about corsair and 30 second long timestamps. I guess CKB causes this, I'm not sure.

    [    2.513966] clocksource: Switched to clocksource tsc
    [   11.049768] hid-generic 0003:1B1C:1B20.0002: usb_submit_urb(ctrl) failed: -1
    [   11.049787] hid-generic 0003:1B1C:1B20.0002: timeout initializing reports
    [   11.049987] input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1B1C:1B20.0002/input/input7
    [   11.105925] hid-generic 0003:1B1C:1B20.0002: input,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-2/input1
    [   21.105817] hid-generic 0003:1B1C:1B20.0003: timeout initializing reports
    [   21.106014] hid-generic 0003:1B1C:1B20.0003: hiddev0,hidraw3: USB HID v1.11 Device [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-2/input2
    [   23.882452] random: nonblocking pool is initialized
    [   31.105803] hid-generic 0003:1B1C:1B20.0004: usb_submit_urb(ctrl) failed: -1
    [   31.105810] hid-generic 0003:1B1C:1B20.0004: timeout initializing reports
    [   31.106001] hid-generic 0003:1B1C:1B20.0004: hiddev0,hidraw4: USB HID v1.11 Device [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-2/input3

And later:

    [   32.478328] input: ckb1: Corsair STRAFE RGB Gaming Keyboard as /devices/virtual/input/input17
    [   32.478638] input: ckb1: Corsair STRAFE RGB Gaming Keyboard as /devices/virtual/input/input18

My theory is that the kernel gives up on finding a driver for the keyboard after 30 seconds, and later CKB kicks in.

I could be wrong, and this could just be my keyboard acting up.

tatokis, 2016-01-31:
From the readme:
You have to set specific kernel paremeters using GRUB so that the hid driver is disabled for the keyboard.

Since you have a Strafe RGB:
```
Strafe RGB: usbhid.quirks=0x1B1C:0x1B20:0x408
```
For more info on how to use it, read the readme.

SicVolo, 2016-01-31:
I have that problem too and I have the quirks on (in fact that's the only way to get Strafe RGB work on 4.2 kernel). In my case it seems that the keyboard messes up the EFI's timing and/or EFI's internal USB device drivers somehow, making the whole boot slower, even before it hits Grub. Grub menu is very slow too.
@kimamb - does your system boot slower BEFORE you see the Grub screen? Do you have the same in NATIVE windows? Did you check with corsair?

tatokis, 2016-01-31:
@SicVolo I see what you mean.
However in this case, since the kernel complains about timeouts in the hid-generic driver, it is most likely the lack of quirks.
If the quirk was enabled, hid-generic wouldn't have tried to initialise the reports in the first place, if I am not mistaken.

That and I would also like to point out that the 0x408 value didn't work for me on my K70 RGB, I had to use 0x20000000 instead.
Specifically in hid.h on the linux kernel source, 0x20000000 is defined as HID_QUIRK_NO_INIT_REPORTS, so maybe this could also help.

kimamb, 2016-01-31:
@SicVolo It does. I thought it was just GRUB messing up, but then I saw in its config file that it is set at no delay.

kimamb, 2016-01-31:
@tatokis I have no idea what quirks are; are they boot parameters?

tatokis, 2016-01-31:
@kimamb yes, they are boot parameters.
You open up /etc/default/grub, and in GRUB_CMDLINE_LINUX you add usbhid.quirks=0x1B1C:0x1B20:0x408 or usbhid.quirks=0x1B1C:0x1B20:0x20000000.
(make sure it is uncommented)

So the final line would be something like:
``` 
GRUB_CMDLINE_LINUX="usbhid.quirks=0x1b1c:0x1b13:0x408"
```
And then you save, run sudo update-grub for grub to be updated and reboot.
What this does is tell hid-generic to ignore reports for your keyboard so that you do not get this long delay due to timeouts.

kimamb, 2016-01-31:
@tatokis 
Didn't seem to work

http://pastebin.com/FsV9JDeg

tatokis, 2016-01-31:
@kimamb Did you also try usbhid.quirks=0x1B1C:0x1B20:0x20000000?
Also, make sure the quirks paremeter is actually appended to the kernel line. (Should be seen in the begining of dmesg.)

If neither of those work, then I do not know, sorry.

kimamb, 2016-01-31:
@tatokis Oh damn. The latter one really helped. 17 sec boot time now.

tatokis, 2016-01-31:
@kimamb This is probably the best you can get for now.
Since you said that you also get slowdowns before GRUB (and assuming while booting Windows as well), you are probably also affected by the issue @SicVolo mentioned.

Unfortunately I can't really help with that. Maybe a UEFI update?

kimamb, 2016-01-31:
@tatokis Grub somehow seems to boot faster as well. Maybe it's placebo. But when rebooting my monitor now goes dark while grub is there, which is normal.

tatokis, 2016-01-31:
@kimamb Could always measure the bootup time with and without the keyboard and then compare.

kimamb, 2016-01-31:
@tatokis seems like too much work. But I/you could compare the bootlogs.

http://pastebin.com/RTsLi2Rw

tatokis, 2016-01-31:
@kimamb The boot logs are from the kernel though, not from UEFI and GRUB, so it doesn't provide much info.
If you don't mind the current bootup time, then there is no point in measuring it.

Anyway, I think the readme should be modified to include 0x20000000 as a solution for "timeout initializing reports", as the current value (0x408) doesn't resolve this issue.

ccMSC, 2016-02-01:
Interesting. I actually had the opposite problem on my system - `0x20000000` didn't fix long boot times, but `0x408` did. It should probably be fine to combine them (`0x20000408`) so I guess I'll update the manual to say that?

kimamb, 2016-02-01:
@ccMSC Yeah, good idea.

tatokis, 2016-02-01:
@ccMSC Agreed.

SicVolo, 2016-02-03:
I changed 0x408 to 0x20000000 and rebooted a couple of times but that did not help with the delays pre and post grub. Did not make anything worse though. I'll try 0x20000408 next.

SicVolo, 2016-02-04:
Tried 0x20000408, the boot AFTER grub is fast, but before grub is still slow. Tried an EFI/NVRAM/SMC reset, no luck. It probably has nothing to do with CKB though, but hardware

ccMSC, 2016-02-05:
I'd guess that the BIOS is trying to initialize the keyboard, but timing out. I don't think GRUB has its own USB driver, although I could be wrong.

kimamb, 2016-02-05:
@ccMSC My keyboard works in BIOS, though.

ccMSC, 2016-02-06:
Hmm. No idea, then.

kimamb, 2016-02-06:
@ccMSC I personally don't have a problem with grub either.

For those who have problems with grub; can you please check if the keyboard works in bios.

frickler24, 2016-02-06:
I get a similar effect.
Booting (mint 17.2) just with the K70RGB and the following line in the grub-defaults:
<pre><code>GRUB_CMDLINE_LINUX_DEFAULT="splash usbhid.quirks=0x1b1c:0x1b12:0x20000000 usbhid.quirks=0x1b1c:0x1b11:0x20000408"
</code></pre> results in fast boot (about 26sec).
With K70RGB + M65RGB and the same line, boot results in 30sec timeouts for the mouse only. KB is OK.
Changing the quirks-param for the mouse either to 408 or 20000408 does not change anything.
Interesting effect is: when doing a reboot, the mouse timeout delays shutting down the kernel also.
BTW: The wait time is only in the kernel, not in grub.

My Question: I found the 0x20000000 as
<pre><code>#define HID_QUIRK_NO_INIT_REPORTS		0x20000000</code></pre> in hid.h, but where do you get the 0x408 from?

tatokis, 2016-02-07:
@frickler24 It's a combination of 0x00000400 and 0x00000008.
You just add them up

frickler24, 2016-02-07:
@tatokis ok, sorry, I had a look into the wrong header-file.
@kimamb Now I have correct parameters for fast booting with both K70RGB and M65RGB.
The KB needs a value 20000408, because the 408 prevents from stopping the driver when using G-keys. For the mouse the 20000000 is needed only.
Then I had some trouble with grub syntax. No help found in google, but the correct syntax seems to be: <pre><code>GRUB_CMDLINE_LINUX_DEFAULT="splash usbhid.quirks=0x1b1c:0x1b11:0x20000408,0x1b1c:0x1b12:0x20000000"</code></pre>
With this, I get "super fast booting" again:
<pre><code>[    1.864937] usb 3-3: Product: Corsair K95 RGB Gaming Keyboard
[    1.864938] usb 3-3: Manufacturer: Corsair
[    1.864939] usb 3-3: SerialNumber: 0F022014AE3B9406537275B6F5001945
[    1.866083] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1B1C:1B11.0003/input/input8
[    1.866273] hid-generic 0003:1B1C:1B11.0003: input,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-3/input0
[    1.866964] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:1B1C:1B11.0004/input/input9
[    1.867081] hid-generic 0003:1B1C:1B11.0004: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-3/input1
[    1.867721] hid-generic 0003:1B1C:1B11.0005: hiddev0,hidraw4: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-3/input2
[    1.868366] hid-generic 0003:1B1C:1B11.0006: hiddev0,hidraw5: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-3/input3
[    1.967588] ata4: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
... bla bla from other devices ...
[    2.332106] usb 3-12: new full-speed USB device number 5 using xhci_hcd
[    2.461511] usb 3-12: New USB device found, idVendor=1b1c, idProduct=1b12
[    2.461512] usb 3-12: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.461513] usb 3-12: Product: Corsair M65 RGB Gaming Mouse
[    2.461514] usb 3-12: Manufacturer: Corsair
[    2.461515] usb 3-12: SerialNumber: 0D02303DAE3D8C69539981F7F5001941
[    2.462444] input: Corsair Corsair M65 RGB Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.0/0003:1B1C:1B12.0007/input/input10
[    2.462703] hid-generic 0003:1B1C:1B12.0007: input,hidraw6: USB HID v1.11 Mouse [Corsair Corsair M65 RGB Gaming Mouse] on usb-0000:00:14.0-12/input0
[    2.463319] input: Corsair Corsair M65 RGB Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.1/0003:1B1C:1B12.0008/input/input11
[    2.463536] hid-generic 0003:1B1C:1B12.0008: input,hidraw7: USB HID v1.11 Mouse [Corsair Corsair M65 RGB Gaming Mouse] on usb-0000:00:14.0-12/input1
[    2.464011] hid-generic 0003:1B1C:1B12.0009: hiddev0,hidraw8: USB HID v1.11 Device [Corsair Corsair M65 RGB Gaming Mouse] on usb-0000:00:14.0-12/input2
[    2.464531] hid-generic 0003:1B1C:1B12.000A: hiddev0,hidraw9: USB HID v1.11 Device [Corsair Corsair M65 RGB Gaming Mouse] on usb-0000:00:14.0-12/input3
[    2.556519] Switched to clocksource tsc
... bla bla again
[   22.513630] vboxpci: IOMMU not found (not registered)
</code></pre> and system is up.
Maybe helpful for others - add it to the README?

tatokis, 2016-02-07:
```Now I have correct parameters for fast booting with both K70RGB and M65RGB.```
I think you meant K95 RGB.

It looks like 0x20000000 is required to fix the report timeout issues, and the 0x408 is required for G-Keys (on keyboards that have them).
Since the K70 RGB does not have any G-Keys, 0x408 didn't change anything for me.

I should try 0x20000408 on my K70 and see if it affects anything.
If my system still boots up as fast as with 0x20000000, then it can probably be used as a universal solution for all the keyboards, whether they have extra keys or not.

ccMSC, 2016-02-08:
`0x400` (`HID_QUIRK_ALWAYS_POLL`) is the flag which makes the G-keys work correctly. IIRC it's also needed for some (but not all) of the mice. `0x8` is `HID_QUIRK_NOGET` which should prevent some of the initialization packets.

I'm not clear on the exact difference between `0x20000000` and `0x8` but it seems that combining both of them should work for everyone.

frickler24, 2016-02-17:
@tatokis yes, I meant K95RGB. Thanks and sorry for the irritation.

zurfyx, 2016-03-20:
K70RGB is indeed 0x20000000
Anyone else having this issue feel free to use this bash script (ready for K70RGB): https://gist.github.com/zurfyx/ba17b7c79d288a79ff07

Other devices might need different values (such as 0x408)

kimamb, 2016-04-21:
After upgrading from Ubuntu 15.10 to Debian 8, the issue reappeared.

usbhid.quirks=0x1B1C:0x1B20:0x408, usbhid.quirks=0x1B1C:0x1B20:0x20000000 nor usbhid.quirks=0x1B1C:0x1B20:0x20000408 works.

BraisFreire, 2016-04-26:
Hi guys.

I'm havins issues with the K70 RGB too. I have Ubuntu 16.04 / W10 and grub. 
Randomly, I can use the keyboard on grub some times, but others not. Inside the OSs I always can use it without problem.
Any ideas of this? I don't know if it is related with all the things you said above.

Thank you son much!

ccMSC, 2016-04-26:
I think GRUB has more to do with the BIOS than with the Linux configuration. Could be wrong though. In either case there's nothing I can do about that AFAIK.

kimamb, 2016-04-26:
If your keyboard has a BIOS switch on it, use it in grub.

closed 2016-06-13 by ccMSC

light2yellow, 2017-05-08:
@ccMSC if you ever return and read this, please let us now how did you deduce this statement:

> 0x400 (HID_QUIRK_ALWAYS_POLL) is the flag which makes the G-keys work correctly. IIRC it's also needed for some (but not all) of the mice. 0x8 is HID_QUIRK_NOGET which should prevent some of the initialization packets.

I'm very curios. Did you just randomly experimented or there's some knowledge behind it? In case of the latter, please explain.

kimamb, 2017-05-08:
CKB is dead.

CKBnext is where you should ask this.

frickler24, 2017-05-09:
@kimamb This questions comes from ckb-next to ckb. All is Ok.
@light2yellow I remember, we had a long discussion about this: 0x400, 0x8, 0x2000000 were the parameters we were testing around with. Eg my quirks are 0x20000408 for both K95RGB and M65. 
`usbhid.quirks=0x1B1C:0x1B11:0x20000408,0x1B1C:0x1B12:0x20000408`
What is the reason for your question?