ccMSC/ckb #235: Can't make

M5oul, 2016-01-31:
On Fedora 23, make fail,
```bash
ckb-0.2.3]$ make
cd src/ckb-daemon/ && ( test -e Makefile || /usr/bin/qmake-qt5 /home/moul/projects/ckb-0.2.3/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && make -f Makefile 
make[1]: Entering directory '/home/moul/projects/ckb-0.2.3/src/ckb-daemon'
make -f Makefile.Debug
make[2]: Entering directory '/home/moul/projects/ckb-0.2.3/src/ckb-daemon'
gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/moul/projects/ckb-0.2.3/src/ckb-daemon/../../VERSION`\" -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o debug/device.o device.c
In file included from includes.h:4:0,
                 from command.h:4,
                 from device.c:1:
os.h:30:21: erreur fatale: libudev.h : Aucun fichier ou dossier de ce type
compilation terminée.
Makefile.Debug:270: recipe for target 'debug/device.o' failed
make[2]: *** [debug/device.o] Error 1
make[2]: Leaving directory '/home/moul/projects/ckb-0.2.3/src/ckb-daemon'
Makefile:38: recipe for target 'debug' failed
make[1]: *** [debug] Error 2
make[1]: Leaving directory '/home/moul/projects/ckb-0.2.3/src/ckb-daemon'
Makefile:49: recipe for target 'sub-src-ckb-daemon-make_first' failed
make: *** [sub-src-ckb-daemon-make_first] Error 2
```

ccMSC, 2016-02-01:
You're missing the development version of `libudev`. I'm not aware of the package name on Fedora, but searching for "udev" should help you find it.

M5oul, 2016-02-02:
Thanks it works with `systemd-devel` package.

closed 2016-02-02 by M5oul