ccMSC/ckb #236: M65 RGB mice issue?

M5oul, 2016-02-02:
I am using a M65 RGB mice, is there an issue?:
```bash
$ sudo ./bin/ckb-daemon 
Mot de passe [sudo] de moul :
    ckb: Corsair RGB driver beta-v0.2.3
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair M65 RGB Gaming Mouse at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usbrecv (via profile_mouse.c:17): Connection timed out
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
```
I can modifiy color and bidings via the interface.

ccMSC, 2016-02-03:
Nope, that's normal. It just means the driver can't read the lighting profile stored in hardware. Sometimes it works, but often it doesn't. Seems to be a hardware issue. It won't affect the driver's functionality.

closed 2016-02-03 by ccMSC

M5oul, 2016-02-03:
Ok, thanks.