ccMSC/ckb #237: Cannot control daemon while ckb application is running

djsmiley2k, 2016-02-06:
When the ckb application controlling animations is running, I cannot pass arguments to the daemon (to change modes, turn off colours etc.)

Once ckb application is quitted, these then work correctly again.

Ideally I'd have a command line way of 'disabling' the ckb application, and then re-enabling it once I wish for it to take over running of animations again. (I'm attempting to script turning off the leds when my screen is locked).

ccMSC, 2016-02-06:
The GUI intentionally overwrites all settings in order to keep its state consistent. A CLI interface that will allow you to control both of them is planned eventually.

In your case, you could auto-stop/restart it, like:
```
# When screen is locked
killall ckb
echo rgb 000000 > /dev/input/ckb1/cmd

# ...

# When screen is unlocked
ckb --background
```

djsmiley2k, 2016-02-06:
Hi ccMSC,

I did consider that, but when I 'stop' ckb application sometimes the
keyboard is reverting to the hardware saved setting, which is a completely
different colour. I've tried save to hardware as well as using hwsave but
no luck so far.

I'll continue to test and raise a new issue if i can't get it started.

closed 2016-02-06 by djsmiley2k

ccMSC, 2016-02-08:
Sorry, I forgot that the keyboard reverts to hardware mode after the GUI is closed.

Using `echo active rgb 000000 > /dev/input/ckb1/cmd` should do the trick.