ccMSC/ckb #239: Edge Case. Using USB Switch now causes driver to malfunction upon switch-back to OSX.

silverjerk, 2016-02-09:
I was using v0.2.2 without issue. However, when moving to v0.2.4, switching mice and keyboards from the PC back to the Mac using a USB Switch causes the settings screen to disappear, and illumination, keybinds and other settings are lost. Unplugging/plugging in the USB cables/switch, and restarting the application doesn't fix the issue -- a full, hard restart of the Mac is required to address the problem. Attached, you can see the configuration screen is all but lost. 

[image]

shazron, 2016-02-09:
Not sure if related, but see #225
Usually with the ckb app open, I just do a `sudo killall -9 ckb-daemon` to reset things.

silverjerk, 2016-02-10:
Thanks. It appears as if we have the exact same use case and running into the same issue. I'm not sure I agree with the other user in the thread suggesting it's impossible, as it switches and works fine in v0.2.2. And the suggestion to use Software switching simply won't work for me. I dev on my Mac, game on my PC, and latency and other concerns would prevent me from using this workaround. 

For now, I'll use your suggested fix, since I always have iTerm2 open and a simple addition to a-text would make the process of resetting really simple.

ccMSC, 2016-02-11:
Could you provide me with a log from both v0.2.4 and v0.2.2 daemons? You can get them by doing this in a terminal:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```

then use the switch and look at the output.

As far as I know you're the only person who's ever got it working with a USB switch, so I'm curious what (if anything) changed between the two versions. I can't promise a fix, as it may be the case that the changes were necessary in order to fix boot issues, but it's worth looking into.

silverjerk, 2016-02-11:
I'm wrapped in a deployment at the moment, but I'll get you the data tomorrow. Right now, my workaround is a little more complex, but working beautifully for v0.2.4. I'm using ControlPlane to listen for the Strafe to be plugged in via USB, after which point it fires a small shell script I wrote to reset the driver. There's an extra 2-3 seconds of delay after the switch is turned back to the Mac from the PC, but it does the trick and keeps me from having to do it manually in the terminal every time.

I did test all the way back from v0.2.4 to v0.2.2, so I can tell you the functionality went haywire with the v0.2.3  update-- so that's likely your ground zero for where the change happened.

silverjerk, 2016-02-19:
Apologies @ccMSC, I was wrapped up with several things this week and was unable to address this. 

@shazron I was able to create somewhat of an elegant workaround. I'm not sure how familiar you are with ControlPlane. I find it indispensable. If you don't already have it, grab it; it's a small download. After you have it installed, open the preferences and create a context titled "Switch." Under Evidence Sources, make sure you have "Attached USB Device" enabled. From there create a new rule, "Add Attached USB Device Rule" and find the Corsair STRAFE RGB on the list of devices. Set the context for this rule to "Switch" (which you created before). Then, create an action to "Open a file" under Finder Actions, and navigate to your .command file containing your shell script for reseting the driver. I went a step further and added an Apple Script function to close the terminal window once the command is executed (this works for me because I use iTerm and never have Terminal open regularly). 

This is my shell script: 

`#!/bin/bash
sudo killall -9 ckb-daemon
osascript -e 'tell application "Terminal" to quit' & exit`

This should open a window, reset the driver, and immediately close the window. It's nearly seamless, and since @ccMSC is stating there may be no fix in place, it would work as a good long term solution for the few cases where users are utilizing USB Switches.

shazron, 2016-02-23:
thanks @silverjerk I will try it out!

closed 2016-06-13 by ccMSC