ccMSC/ckb #240: no keyboard on restart

2-alex-2, 2016-02-11:
start by saying great work so far has aloud me to use a great keyboard on my hackintosh. so on restart on my hackintosh doesn't recognise the keyboard and i need to unplug and replug in the keyboard for the keys to work but also cob doesn't recognise it even after i unplug and replug in the keyboard. have you had anything like this ? 

thanks alex

Br0ckout, 2016-02-11:
I get this issue too (K70 RGB). Sometimes it will detect the keyboard and sometimes it wont. I find that when it doesn't detect the keyboard, OSX takes longer to boot (OSX loading bar slugs along). I've tried different USB ports both 2.0 and 3.0 but it makes no difference.

ccMSC, 2016-02-12:
See #233

closed 2016-02-12 by ccMSC