ccMSC/ckb #241: HighDpi monitor ui positioning problems.

mattanger, 2016-02-12:
Hi, I've got a monitor with a 3200x1800 resolution and 235 ppi. I've recently purchase a k70 RGB and I this project is awesome! But! The UI is a bit messed up for me. Specifically, lighting tab is off some, not too serious though. The most bothersome is the binding widget in the binding tab. All of the options are cut off/stacked too closely or whatever. 

I've forked and started digging through some of the UI code, and started making some changes to the kblightwidget.ui to make it look nicer, for me at least. I was planning on going through in working on some of the other widgets as well. My question is, am I missing something? Or does HighDpi support need to be worked on? If so I'm willing to start working on this!

ccMSC, 2016-02-15:
Qt recently made changes to their HiDPI support, which might be the cause of this. Does it affect any other apps that you know of?

For now, try running ckb with `QT_DEVICE_PIXEL_RATIO=2 ckb` and see if that makes a difference.

mattanger, 2016-02-15:
I'll give that a shot. I've come across several QT apps with the same issue, but I can't think of a really good example right now. I'll update when I think of one.

closed 2016-06-13 by ccMSC