ccMSC/ckb #243: No Devices Connected

damccoard, 2016-02-14:
So I have the Scimitar RGB Mouse. I downloaded the ckb app and ran it. The app detected the mouse after a few seconds and I could program it like normal. The Corsair program did a update and now ckb will not detected the mouse. I have  fallow some of the troubleshoots but nothing is working. this it wat the terminal is showing when I open it.
Last login: Sun Feb 14 00:42:17 on console
/Applications/ckb.app/Contents/MacOS/ckb ; exit;
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-gradient"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-pinwheel"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-rain"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-random"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-ripple"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-wave"
qt.network.ssl: QSslSocket: cannot resolve SSL_set_psk_client_callback
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_server_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSL_select_next_proto
qt.network.ssl: QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb
qt.network.ssl: QSslSocket: cannot resolve SSL_get0_next_proto_negotiated
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get0_next_proto_negotiated
No GPG detected, signature verification disabled
Downloaded new firmware list. 6 entries found.

ccMSC, 2016-02-15:
Edited title for clarity. It looks like the new firmware isn't compatible. I'll try to get the driver updated for it ASAP.

ccMSC, 2016-02-28:
As an update to this, I have a Linux version available for the new firmware, but I've been unable to get it working with the OSX version and I may have to rewrite part of the driver. It may be another week or more before I can get a release ready.

ccMSC, 2016-03-14:
OSX update is ready for testing. I've posted the changes in the `fw2` branch.

Here's a binary which should correctly work with the new firmware version: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.4%2Bf02.pkg

tjw0lf, 2016-03-29:
Please don't close my thread without letting me know first please. as its not solved for me
What are the step and everything i have to do? Where is it Installation steps
Please thank you

tjw0lf, 2016-03-31:
OK so i installed it and just run the .quickinstall and that was it. Thank you very much for the fix ccMSC awesome job.

Just for future reference can you be more specific on how it needed to be installed or upgraded. 
I was looking at the readme file and it said to remove ckb and install. or remove certain files then do quickinstall.
I just guess and run quickinstall and worked fine. But would have been easier if you told me thats all needed to be done. Wasn't sure

Also I'm not sure if this is a bug or if you just had it set to this. But when you first install this new version the light is green on the DPI. But when in windows the default color is white. SO lunching the ckb program software etc. by default its not white and you can not change the color of the DPI. But in performance if i unchecked DPI  indicator you can change the color to white or whatever you like. Me i like it whiote the basic color of corsair. But does this need to be checked or what is the point of this being checked and why not can you change this color when it is checked. I don't understand this feature.

Please let me know if this is right or still being worked on or what does the DPI indicator do  as you can not change the color when checked. its green

Thank you

ccMSC, 2016-04-01:
The fw2 branch is here: https://github.com/ccMSC/ckb/tree/fw2

However it's not likely to see any more updated, at least not for Linux. It's going to be moved back into the main branch as soon as I solve a few remaining issues on the OS X side.

When the DPI indicator is active, there's a different color setting for each individual DPI level. Level 3 (the middle setting) is green by default. If you click on the color icon near the left, you can change it to some other color. When you use the DPI Up or DPI Down buttons on the mouse the indicator will change according to which level is selected.

closed 2016-04-08 by ccMSC