ccMSC/ckb #244: Suspend to RAM and wake up

Tazmain, 2016-02-14:
After suspending to ram and waking up the pc, the lights on the keyboard stay frozen and the system does not detect key presses. If you unplug the keyboard and plug it back in ( after using another keyboard to login again) the ckb program does not detect a keyboard. Happening with K95 RGB.

ccMSC, 2016-02-15:
Seems most likely to be a hardware issue. Have you tried a different USB hub?

damccoard, 2016-02-15:
It dose that with the Corsair Utility Engine, all you have to do it restart the program.

Tazmain, 2016-02-19:
In windows you can still use the keyboard. 

So I took an old keyboard plugged it into the USB port I have the Corsair in, Plugged the Corsair into a different port. First of all the ckb-daemon had to be restarted to pick up the keyboard because ckb showed no keyboard. I then suspened to ram, and woke up the pc. Yet again the lights where frozen and the Corsair was not usable. My old keyboard in the port I usually have my Corsair in was working just fine. Once logged in, I open ckb, it showed no keyboard and the daemon took a while to restart. Once it was restarted then only did my Corsair work again. So it does not seem to be a USB hub issue nor hardware.

Tazmain, 2016-02-19:
@damccoard -> Corsair Utility Engine, in windows does not do this. I put my pc to sleep, let it power off all devices like it does with linux, restored full power and as soon as I saw the windows 10 login screen the keyboard was fully functioning with animated lightening

damccoard, 2016-02-19:
I have Window's 7 so that might be the reason it dose it with me.

frickler24, 2016-02-19:
@Tazmain  with linux Mint 17.1 and 17.2 suspend-to-ram and powering on again is working fine. What distribution do you use?

tatokis, 2016-02-20:
I seem to be having a similar issue as well.
When I suspend, the keyboard light animation pauses for a moment and then the lights go off completely.
The computer wakes up normally by pressing a key on my K70 RGB.

The Unity login screen shows up, however the keyboard LEDs are no longer animating (frozen in the same state as before suspending), and no keytrokes go through.
Everything else is running normally while this is happening, mouse moves and it is possible to log in using another keyboard.

After some time, the keyboard resumes normal operation.
Running dmesg reveals this:
```
[195356.220117] PM: resume of devices complete after 3079.697 msecs
[195356.220515] usb 8-1:1.0: rebind failed: -517
[195356.220519] usb 8-1:1.1: rebind failed: -517
[195356.220523] usb 8-1:1.2: rebind failed: -517
[195356.220526] usb 8-1:1.3: rebind failed: -517
[195356.221193] PM: Finishing wakeup.
[195356.221196] Restarting tasks ... 
[195356.230492] usb 8-1: usbfs: process 1558 (ckb-daemon) did not claim interface 0 before use
[195356.230510] usb 8-1: usbfs: process 1558 (ckb-daemon) did not claim interface 1 before use
[195356.230564] usb 8-1: usbfs: process 1558 (ckb-daemon) did not claim interface 2 before use
[195356.230655] done.
[195356.250976] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-1/8-1:1.0/0003:1B1C:1B13.0008/input/input27
[195356.307477] hid-generic 0003:1B1C:1B13.0008: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-1/input0
[195356.315996] hid-generic 0003:1B1C:1B13.0009: hiddev0,hidraw2: USB HID v1.11 Device [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-1/input3
[195356.878959] usb 8-1: usbfs: process 1148 (ckb-daemon) did not claim interface 3 before use
[195357.090457] usb 8-1: reset full-speed USB device number 2 using xhci_hcd
[...]
[195372.194689] usb 8-1: device descriptor read/64, error -110
[195387.417340] usb 8-1: device descriptor read/64, error -110
[195387.637612] usb 8-1: reset full-speed USB device number 2 using xhci_hcd
[195402.756060] usb 8-1: device descriptor read/64, error -110
[195417.978633] usb 8-1: device descriptor read/64, error -110
[195418.202723] usb 8-1: reset full-speed USB device number 2 using xhci_hcd
```
The last usb reset is when the keyboard resumes functioning.

However, if the ckb application is closed (and the daemon is still running), the keyboard functions normally after waking up.

Output of dmesg without the ckb application running:
```
[195898.300351] PM: resume of devices complete after 3011.713 msecs
[195898.300717] usb 8-1:1.0: rebind failed: -517
[195898.300721] usb 8-1:1.1: rebind failed: -517
[195898.300725] usb 8-1:1.2: rebind failed: -517
[195898.300729] usb 8-1:1.3: rebind failed: -517
[195898.301379] PM: Finishing wakeup.
[195898.301382] Restarting tasks ... 
[195898.301806] usb 8-1: usbfs: process 1558 (ckb-daemon) did not claim interface 1 before use
[195898.301867] usb 8-1: usbfs: process 1558 (ckb-daemon) did not claim interface 2 before use
[195898.307465] done.
[195898.331311] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-1/8-1:1.0/0003:1B1C:1B13.000A/input/input30
[195898.387738] hid-generic 0003:1B1C:1B13.000A: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-1/input0
[195898.396077] hid-generic 0003:1B1C:1B13.000B: hiddev0,hidraw2: USB HID v1.11 Device [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-1/input3
```

Running the ckb application afterwards causes usb resets and device descriptor read errors.
Again, after some time, ckb finally recognises the keyboard and resumes normal operation.

I am running Ubuntu 15.10 with kernel 4.5.0-rc4.

@Tazmain please provide logs from dmesg.

Tazmain, 2016-02-20:
@frickler24  I am running on arch linux.

Tazmain, 2016-02-20:
So after using another keyboard to login, and waiting for a bit, I saw some lights flash and finally the keyboard started working agian with animated lightening. 

Dmesg output:
[   99.550037] ehci-pci 0000:00:1a.0: cache line size of 64 is not supported
[   99.550052] sd 6:0:0:0: [sdg] Starting disk
[   99.550088] ehci-pci 0000:00:1d.0: cache line size of 64 is not supported
[   99.863457] usb 5-1: reset high-speed USB device number 2 using xhci_hcd
[   99.873378] ata5: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[   99.873381] usb 1-1: reset high-speed USB device number 2 using ehci-pci
[   99.876715] usb 2-1: reset high-speed USB device number 2 using ehci-pci
[   99.884514] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[   99.884518] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT0._GTF] (Node ffff8804af13e280), AE_NOT_FOUND (20150930/psparse-542)
[   99.894496] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[   99.894499] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT0._GTF] (Node ffff8804af13e280), AE_NOT_FOUND (20150930/psparse-542)
[   99.895285] ata5.00: configured for UDMA/133
[   99.923503] usb 3-1: reset high-speed USB device number 2 using xhci_hcd
[  100.260117] usb 3-6: reset full-speed USB device number 4 using xhci_hcd
[  100.263467] usb 5-1.1: reset full-speed USB device number 3 using xhci_hcd
[  100.413497] usb 5-1.3: reset high-speed USB device number 4 using xhci_hcd
[  100.593451] usb 3-5: reset full-speed USB device number 3 using xhci_hcd
[  100.690083] ata6: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[  100.690403] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[  100.690424] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT1._GTF] (Node ffff8804af13e2f8), AE_NOT_FOUND (20150930/psparse-542)
[  100.690762] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[  100.690768] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT1._GTF] (Node ffff8804af13e2f8), AE_NOT_FOUND (20150930/psparse-542)
[  100.690832] ata6.00: configured for UDMA/100
[  100.783480] usb 5-1.3.4: reset high-speed USB device number 5 using xhci_hcd
[  101.336804] usb 5-1.3.4.3: reset low-speed USB device number 6 using xhci_hcd
[  101.627092] usb 5-1.3.4.3: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[  101.627098] usb 5-1.3.4.3: ep 0x82 - rounding interval to 64 microframes, ep desc says 80 microframes
[  101.676999] PM: resume of devices complete after 2131.141 msecs
[  101.678083] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:1B1C:1B11.000A/input/input30
[  101.730373] hid-generic 0003:1B1C:1B11.000A: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-5/input0
[  103.250087] ata3: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  103.420306] ata3.00: configured for UDMA/133
[  104.893401] ata7: link is slow to respond, please be patient (ready=0)
[  104.896731] ata8: link is slow to respond, please be patient (ready=0)
[  104.903393] ata1: link is slow to respond, please be patient (ready=0)
[  104.906724] ata2: link is slow to respond, please be patient (ready=0)
[  104.913391] ata4: link is slow to respond, please be patient (ready=0)
[  105.426738] ata7: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  105.464639] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[  105.464646] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT2._GTF] (Node ffff8804af13e370), AE_NOT_FOUND (20150930/psparse-542)
[  105.490074] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  105.540059] ata8: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  105.571820] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[  105.571827] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT3._GTF] (Node ffff8804af13e3e8), AE_NOT_FOUND (20150930/psparse-542)
[  105.643837] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[  105.643845] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT2._GTF] (Node ffff8804af13e370), AE_NOT_FOUND (20150930/psparse-542)
[  105.643971] ata7.00: configured for UDMA/133
[  105.709944] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
[  105.709952] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT1.SPT3._GTF] (Node ffff8804af13e3e8), AE_NOT_FOUND (20150930/psparse-542)
[  105.710085] ata8.00: configured for UDMA/133
[  105.710675] ata1.00: configured for UDMA/133
[  106.300066] ata4: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[  106.324430] ata4.00: configured for UDMA/133
[  107.733405] ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  107.875772] ata2.00: configured for UDMA/133
[  111.730074] hid-generic 0003:1B1C:1B11.000B: usb_submit_urb(ctrl) failed: -1
[  111.730084] hid-generic 0003:1B1C:1B11.000B: timeout initializing reports
[  111.730630] input: Corsair Corsair K95 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.1/0003:1B1C:1B11.000B/input/input31
[  111.783569] hid-generic 0003:1B1C:1B11.000B: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-5/input1
[... cut ...]
[  328.520076] hid-generic 0003:1B1C:1B11.0011: usb_submit_urb(ctrl) failed: -1
[  328.520099] hid-generic 0003:1B1C:1B11.0011: timeout initializing reports
[  328.520292] hid-generic 0003:1B1C:1B11.0011: hiddev0,hidraw8: USB HID v1.11 Device [Corsair Corsair K95 RGB Gaming Keyboard ] on usb-0000:00:14.0-5/input3
[  328.777504] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input34
[  328.778222] input: ckb1: Corsair K95 RGB Gaming Keyboard as /devices/virtual/input/input35

frickler24, 2016-02-20:
@Tazmain There is another issue documented about using quirks  when booting linux with the K95. In your log I can see the same timeouts as I had before using the quirks. The timeout is about 30 seconds, so it might fit to your description.

Tazmain, 2016-03-28:
Well thing is that at times the daemon also makes my system take long to shutdown because it's stuck on stopping the daemon, But I will have a look into the quirks.

closed 2016-06-13 by ccMSC