ccMSC/ckb #245: suggestions

tvalleau, 2016-02-15:
Hello

I'm the guy who worked with Keyboard Maestro (KM) to get the G18 macro key extensions working with KM, and started the thread on the Corsair forum about it.

Here are my comments on this software v.24

1) thank you for your efforts !

I use the G18 extended keys a lot, so ... 

2) the G18 keys, when unbound, should continue to pass through the key identifier, instead of just doing nothing at all. That way the device key (G1, G2...G18) would (continue to) be recognized as something happening (and continue to work with KM.)

3) I tried to work around that with the CKB software, but it doesn't allow multiple modifiers. For example, I was going to rebind G1 to Shift-Command-Tab, but could not because on one of the modifiers can be chosen. Please enable multiple modifiers.

4) There seems to be no way AFAICS to have one of the three keyboard memories switch in when a specific application is launched. I have M1 set up as a general lighting layout; M2 set up for Photoshop and M3 set for FCP. IMHO it should be possible to have the driver switch between these lighting layouts when Photoshop or FCP launches, and revert to G1 when the app quits.

Those are what I see as the biggest limitations so far, FWIW.

And, while I'm here, let me ask if you'd be willing to share the information/code/keyboard-toggle necessary for #4 to work from within Keyboard Maestro? I'd be delighted to pass it along to Peter at Stairways, so that he could implement auto switching. (Well, so that I could provide it, and -ask- him politely if he would implement it. (As a retired programmer myself, I suspect it's not difficult to do. Something like "tell usb device ###, choose mem 2" or something similar.)

Thanks again for all your efforts, and once these issues are resolved, I'll be happy to pay for the software. 

Cordially

tvalleau

Rabcor, 2016-02-16:
About using multiple modifiers, I've been using xdotool for macros, I agree that it should be possible to make macros within the ckb software (in a way that is not dependent on desktop environment, i.e. would work even in console mode), but for now this works under X and in many cases as well under Wayland (granted you have xwayland).

examples include
`xdotool key super+F3`

For an (obviously) super/meta/command/windows+F3 key macro that happens instantly when you press a key down.

If you want something more elaborate...

On key press:
`xdotool keydown control+Page_Up`

On key release:
`xdotool keyup control+Page_Up`

Allows you to hold down a macro until you release the button.

As for switching M modes when a specific program is focused, that is unrealistic to implement under linux since the commandline option for detecting which program is focused can vary greatly between desktop environments (not to mention display servers now that we have Wayland and Mir in the mix). It's not impossible but it shouldn't be a high priority development.

tvalleau, 2016-02-16:
First, my thanks for responding.

As to switching M modes, Keyboard Maestro has been doing that, and quite reliably, for some number of years now. In fact, many Mac OSX programs switch their attributes when the focused app changes. Another example of that is Wacom tablets, which allow for different settings depending on the front-most app. Kensington Trackball works is another, as is the Contour Shuttle pro... the list goes on.

Again, however, my thanks for taking the time to reply.

Rabcor, 2016-02-17:
Ah that's on macs, I was talking about Linux. On macs this is easy, since macs only have one official windowing system. Yes it is indeed possible on OS X, but not very feasible on Linux.

I'm not sure if the lead developer of this project is interested in doing this or not (I somehow get the impression he mostly uses linux himself) but since it can be done on macs, someone simply needs to delve into the code and find out a way to implement this. Surely there's a guide somewhere online how to detect the focused app on macs somewhere on the net, and this can be implemented by mixing that with the programs existing code for switching M modes. The only problem this may bring to the table is the GUI edits which will most likely result in the OS X GUI needing to be different from the Linux one. Doesn't sound that hard to do though, if I used macs at all (personally I despise OS X, so not gonna happen) I'd possibly have done this myself already. I've done similar things in the past.

tvalleau, 2016-02-17:
Thanks for taking the time to reply. (Finding the focused app is a 
trivial toolbox call in OSX.) It can even be done from AppleScript:

tell application "System Events"
	set applicationName to name of every process whose frontmost is true 
and visible ≠ false
end tell

What I'm really looking for, (without having to dig into his code, as 
there is no assurance at all that it is even there), is the hook and 
command to send to the device to get it to select M1, M2 or M3.

KM can easily determine the app with focus, and now just needs to send 
the appropriate command to the K95 to get it to switch.  I just don't 
know what that command is.

Once again, my thanks for your courteous reply.

Tracy
www.valleau.gallery

frickler24, 2016-02-17:
@tvalleau I did some development for creating macros via G1..18.
Integration in the testing path will still need some more time, as ccMSC told me. But you may may have a look at the version I sent a pull request for (branch macro-client.0.0.2).

Using the G-keys with modifiers is still a problem, because the ckb driver acts in an unwanted way.
If you do something like 'echo "macro g1:+f,-f,+o,-o,+o,-o" > your_cmd_pipe; echo "macro lshift+g1:+x,-x,+y,-y,+z,-z" > your_cmd_pipe', you may program the KB (until now tested on K95RGB, not on your Maestro).
Pressing G1, it will send you "foo" - as you wanted. Pressing lshift+G1, you get "fooXYZ".
That's first a problem in the driver. Until now I located the position to change, but have no good idea how to change... Second is the interpretation of the modifier key. For this I have the idea to modify the program-string sent to the keyboard like "lshift+g1:+lshift,-lshift,+x,-x,+y,-y,+z,-z".
With this modification, you get "fooxyz", but changing the macro itself may have other side effects.
In the next weeks I will do some further thinking and development on this.

tvalleau, 2016-02-17:
Thank you Lutz. That information helps me understand the issues 
involved. I'm sure whatever you eventually come up with will make the 
product only that much more useful.

Tracy
www.valleau.gallery

ccMSC, 2016-02-18:
To point 2 - we actually can't. Neither Linux nor Xorg has any concept of macro keys, so there are no sane bindings for the keys by default. Same thing happens if you turn off the driver; they don't register at all.

frickler24's macro client mostly addresses the second point, as it allows you to map multiple actions to one keydown. This is still a work in progress.

Ability to switch profiles on an application basis is planned, but it's not high on the priority list at the moment.

tvalleau, 2016-02-18:
Hello, and thanks for the reply.  I either don't completely understand, 
or I was not clear enough in the setup to all this:

1) I'm talking about Mac OSX, and no other OS.
2) Keyboard Maestro -IS- a macro program. It lets the user set up 
complex sequences of actions, and then trigger them with a 
(configurable) keystroke.
3) without -any driver at all- Keyboard Maestro succeeds in reading and 
recognizing individual G1-18 keys, and so right now IS working to 
trigger one of its macros. (Installing this open-source driver 
-disables- that currently working capability.)
4) what I was asking for was -not- macro capability, since I already 
have that. (#3 above)  I was wondering if someone who knows would share 
what needs to be sent programmatically back to the keyboard, in order to 
invoke a toggle.  If not, does anyone know if Corsair has a low-level 
programming guide for the K95RGB, which would likely contain that 
information. I'm not asking for this as an end-user, but as a programmer 
myself.

Tracy
www.valleau.gallery

frickler24, 2016-02-18:
@tvalleau thank you for the more detailled description. The driver is used on different unix / OSX distributions, therefore the implementation must fit 1-to-all. AFAIK, I do not have a tool like Maestro on my mint distribution. But I have an idea. how to combine the general requirement for using G-keys (let's say: Instructing the keyboard to send free definable keystrokes when pressing Gxx) and the ability to call a program (like your maestro) instead.
In src/ckb/keyaction.cpp is a member function KeyAction::defaultAction(const QString& key) - a large function. 
At the end of that function, you find a line "} else if (prefix == "$macro") {" with a comment beginning with "Do nothing., ..." . 
That might be the point to call an external program or better to trigger the calling. Because I don't own a Mac and do not know anything about available APIs of Maestro to call it, I just can give you that hint.

tvalleau, 2016-02-18:
Thank you for that. I appreciate your courtesy.

Tracy
www.valleau.gallery

frickler24, 2016-02-18:
@tvalleau  It's good old german school ;-) 
I just had an other idea: 
As you mentioned, the KB does not send anything when hitting a G-key without the macro definition facility. But did you check, if you can use the binding via the "program" function in ckb?
With this, you can call a program with parameters directly when hitting a G-key, even if the KB does not send any character.
When opening the Binding / Program tab, you can see a hint to launch a program via xdg-open. I am sure, there is a similar way to do that on OS-X. That may help you.
Lutz

ccMSC, 2016-02-20:
On OSX it's a slightly different situation to Linux. Unfortunately ckb has to stop the OS from talking to the keyboard completely or else it can't function. That being the case, there's no way to pass the macro keys back to the operating system, because the OS doesn't understand them beyond raw USB events (which ckb is not able to create either).

You can't actually toggle the M state on the keyboard itself. In fact you can't control anything other than the lighting. The non-RGB K95 had more programmability in this regard, but for the RGB models, the illusion of mode-switching is accomplished in software.

tvalleau, 2016-02-20:
Aha!... and Bummer   :-(

Well, thanks for listening, and for the unfortunate final answer.

I appreciate it.

Tracy
www.valleau.gallery

closed 2016-04-08 by ccMSC