ccMSC/ckb #246: Keyboard not recognized

vlucendo, 2016-02-16:
Hello,

I'm using ckb with the latest release of El Capitan and sometimes when I boot (or reboot) the keyboard isn't recognized and I can't type my password on the login screen. Unplugging it and pluging again makes the keyboard work, but ckb doesn't recognize it until I restart the program using the terminal.

This happens maybe 3 of 4 times I start the system, so it's very frustrating.

Other than that it's an excellent piece of software and does everything I need it to do, thank you very much for the development :)

Rabcor, 2016-02-16:
Never had this problem on Arch or Fedora. I do know though that the kernel still does not support this keyboard so without CKB i've had to use the following grub commandline setting: `usbhid.quirks=0x1B1C:0x1B11:0x408`

For my Corsair k95-rgb. Btw you left out what keyboard exactly you are using.

2-alex-2, 2016-02-17:
Where did you put that command line into. I'm running a k60 and mine has similar issues.

Rabcor, 2016-02-17:
https://github.com/ccMSC/ckb#troubleshooting

(If using grub, you need to put this in /etc/default/grub on the GRUB_CMDLINE_LINUX line; Can also be entered during boot if you edit the commandline (think you press E on the bootloader screen)

Seems like the k60's code isn't listed. You can find out the first part of the code (e.g. 0x1B1C:0x1B11) 

For example, my keyboard listed in lsusb looks like so:

**Bus 004 Device 004: ID 1b1c:1b11 Corsair** after you have that, just try adding the same 0x408 to the end of the line as I had in my command, might work.

ccMSC, 2016-02-18:
See #233

closed 2016-04-08 by ccMSC