ccMSC/ckb #247: STRAFE RGB not responsive at OS X 10.11 boot | CKB install issue

shanece, 2016-02-16:
I have recently just built my first hackintosh and got the STRAFE RGB keyboard to go with my system. After finally getting my system to successfully boot using UniBeast and MultiBeast, the last thing i had to configure was the keyboard. I downloaded the CKB to my system and with initial launch it appear to work, until a minute in when the program no longer recognized my keyboard. I then did the unplug/plug usb to see if that would work and it did not. I then decided to restart my OS, when rebooted, I find that the keyboard is no longer recognized at mac start up and I have to manually configure the keyboard (everytime it's rebooted) with system preferences. Once I get my system to re acknowledge my keyboard, it would work, but once restarted, the issue presist and I don't have functionality to the keyboard. 

I decided to uninstall the pkg and reinstall it but now I can't. I get this message when i try to do anything with the files.

~ name$ /Users/name/Desktop/ckb-master/quickinstall ; exit;
readlink: illegal option --f
usage: readlink [-n] [file ...]
Preparing build giles...
No suitable qmake found.
Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually

Press enter to exit.

Can somebody walk me thru how to reinstall and bypass this no suitable qmake found?
Also where exactly these changes need to be made, with corresponding commands would be helpful. 
I just relatively new to this. Thanks for any advice in advance.

shanece, 2016-02-16:
UPDATE: I was able to get the CKB package to reinstall the application. Now, the application does not recognize the device is plugged in. Any idea on how to fix this.

SicVolo, 2016-02-17:
Have you tried unpluggin and replugging the keyboard into a different usb port? That helps me sometimes. What's your hackintoshi - OSX running on run off the mill intel hardware, or linux runnning on Apples hardware?

ccMSC, 2016-02-18:
By any chance was the firmware updated recently? Corsair released new firmwares for the everything and ckb isn't compatible yet.

shanece, 2016-02-20:
@SicVolo I switched out the USB port the keyboard was plugged into and that made the keyboard functional but still not recognizable by CKB. 

As for my hackintosh; I'm running El Capitan 10.11 on my system. I am not sure if my OS X is running off of mill hardware. I do have clover installed as my boot loader (if that helps). The processor I'm currently using is Intel i7 4790K, on a Asrock H97M Pro4 MoBo. 

@ccMSc if the firmware was updated, would there be away to back log for the old firmware so it could be compatible

ccMSC, 2016-02-24:
disregard what I said, I was looking at the wrong file. The current STRAFE firmware is in fact compatible with ckb.

Could you give me a log from the driver? You can get it by running these commands in a Terminal:

```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```

nbranch1311, 2016-04-12:
@ccMSC I still can not get my Strafe RGB to work with my 10.11.4 OS X build. I commented on a more relevant thread but I wanted to note it here to. I ran the: 
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
command but the file is not there. Neither is the folder 'LaunchDaemons'

Thank you for any help you can provide

closed 2016-06-13 by ccMSC