ccMSC/ckb #249: Settings don't save when quitting

bduong, 2016-02-17:
If I quit CKB on OSX 10.11, my animation and settings don't get preserved.

Oddly enough my lighting profile does get preserved just not the animation.

I'm using a K70.

ccMSC, 2016-02-18:
Animations don't work without the program running. It's a hardware limitation.

bduong, 2016-02-18:
Sorry should have been more clear,

If I quit CKB and restart it, the animations and settings aren't saved for some reason.

ccMSC, 2016-02-20:
Is there a file named `~/Library/Preferences/com.ckb.ckb.plist`? If so, could you upload it here?

bduong, 2016-02-22:
// !!! BINARY PROPERTY LIST WARNING !!!
//
// The pretty-printed property list below has been created
// from a binary version on disk and should not be saved as
// the ASCII format is a subset of the binary representation!
//
{	"Devices.0902103AAEA7A44454E09FE3F5001944.CurrentProfile" = "{64241966-7B59-4A7A-A578-97344EBD2718}";
	"Devices.0902103AAEA7A44454E09FE3F5001944.Profiles" = "{BA7FC152-2D51-4C26-A7A6-A036CC93D924} {64241966-7B59-4A7A-A578-97344EBD2718}";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Binding.KeyMap" = "K70 US";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Binding.UseRealNames" = :true;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Binding.WinLock" = :false;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.GUID" = "{1BA2AC08-F3A0-4973-85FD-31AD33E3E6CC}";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.HwModified" = "75406def";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.List" = ( "{75921CCB-0C1E-433A-9FF3-8168B07CB177}" );
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.BlendMode" = "Normal";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Keys" = ( "down", "right", "numenter", "colon", "slash", "rwin", "prev", "dot", "prtscn", "tab",
		"num0", "num1", "num2", "num3", "num4", "play", "num5", "num6", "num7", "num8", "ralt",
		"num9", "bslash", "esc", "f1", "f2", "f3", "f4", "rctrl", "f5", "f6", "f7", "f8",
		"f9", "numslash", "up", "grave", "pause", "minus", "numplus", "home", "rbrace", "comma",
		"lctrl", "stop", "rshift", "numdot", "0", "1", "2", "lwin", "3", "4", "5", "6", "7",
		"lbrace", "8", "9", "numminus", "end", "ins", "numstar", "lshift", "lalt", "a", "b",
		"c", "equal", "d", "numlock", "e", "f", "enter", "g", "h", "i", "j", "k", "l", "m",
		"n", "space", "o", "pgdn", "p", "q", "r", "s", "t", "u", "v", "f10", "w", "f11",
		"scroll", "x", "f12", "y", "z", "del", "quote", "pgup", "caps", "left", "rmenu",
		"next", "bspace"
	);
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Name" = "Type lighting";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Opacity" = "1";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.color" = "0:ffff00ff 100:00ff00ff";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.delay" = 0;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.duration" = 3;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kpdelay" = 0;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kphold" = "1";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kpmode" = 0;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kpmodestop" = :false;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kprelease" = :false;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kprepeat" = 3;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kpstop" = 0;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.kptrigger" = "1";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.repeat" = 3;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.stop" = -1;
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.Parameters.trigger" = "0";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.ScriptGuid" = "{54DD2975-E192-457D-BCFC-D912A24E33B4}";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{64241966-7B59-4A7A-A578-97344EBD2718}.0.Lighting.Animations.{75921CCB-0C1E-433A-9FF3-8168B07CB177}.ScriptName" = "Gradient";
[... cut ...]
	"Devices.0902103AAEA7A44454E09FE3F5001944.{BA7FC152-2D51-4C26-A7A6-A036CC93D924}.Modified" = "3c856066";
	"Devices.0902103AAEA7A44454E09FE3F5001944.{BA7FC152-2D51-4C26-A7A6-A036CC93D924}.Name" = "ckb Demo";
	Popups.BGWarning = :true;
	Program.DidLoginItem = :true;
	Program.GlobalRemap.lalt = "lctrl";
	Program.GlobalRemap.lctrl = "lwin";
	Program.GlobalRemap.lwin = "lalt";
	Program.GlobalRemap.ralt = "rctrl";
	Program.GlobalRemap.rctrl = "rwin";
	Program.GlobalRemap.rwin = "ralt";
	Program.KbdLayout = "us";
	UI.Light.ShowBaseOnly = :false;
}

ccMSC, 2016-02-24:
So it looks like the changes were in fact saved to disk, but the app is failing to reload them. I'll see if I can figure out what's causing it.

bduong, 2016-02-24:
It also seems to not be saving that file anymore. If I move the plist file and then go through and configure the app again. It doesn't save a new plist file.

I've checked the permissions on the Preferences folder and I own the directory.

closed 2016-06-13 by ccMSC