ccMSC/ckb #250: [K95]How to eject my superdrive, by a key binding

joinscome, 2016-02-18:
Hi there,

Thank you for this great application. 

I have no issue.

But I would like to be able to open and close my two DVD burners (I have a Mac Pro early 2009), using a key, like it is possible with an Apple keyboard. 
By a key binding with a function key, for example.
Is this possible ?

[image]

ccMSC, 2016-02-20:
The binding tab should display an option for Eject under "Keyboard > Media"

joinscome, 2016-02-20:
Thank you for the quick answer.
This does not work for me.  I selected a key to try that , but nothing happens.

Is there any possibility to just open or close the tray ?

frickler24, 2016-02-21:
Do you have a (shell-) command to open the tray?

joinscome, 2016-02-21:
Yes : 

drutil tray open (or _close_ or _eject_) -drive 1 (or _2_)

The Mac Pro has two DVD burners. Number 2 is for the upper one. and number 1 si for the other (of course...).

tatokis, 2016-02-21:
Under binding, after selecting a key, you can go under the "Program" tab, tick the "Launch program on key press" checkbox, and then enter that command in the text box below.
Should work

joinscome, 2016-02-21:
Thank you. It works.

closed 2016-02-21 by joinscome