ccMSC/ckb #254: Scimitar RGB not connecting

tjw0lf, 2016-02-19:
If you can help on why not connecting please

Would love for it to work

I'm on corsairs forum @ImtheBest

ccMSC, 2016-02-20:
There's an issue with the firmware right now. I'm trying to get it fixed soon.

ccMSC, 2016-03-14:
Sorry, I should have closed this earlier as a duplicate of #243. Potential solution is posted in that thread, try it out and let me know if it's working for you.

closed 2016-03-14 by ccMSC

tjw0lf, 2016-03-29:
Please don't close this without letting me know first please. 
What are the step and everything i have to do? Where is it Installation steps
Please thank you

ccMSC, 2016-03-31:
If you're on OSX, I provided a link to an installer which supports the new firmware: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.4%2Bf02.pkg

On Linux, download the `fw2` branch (zip link: https://github.com/ccMSC/ckb/archive/fw2.zip ) and follow the installation as normal.

CMasterson, 2016-03-31:
New installer fixed the issue for me, thanks!

tjw0lf, 2016-03-31:
Sorry i should have stated I'm running Linux. Not OS X or nothing like that no MAC guy here. And where can i find the fw2 for future references