ccMSC/ckb #255: Option to start ckb minimized on Linux

sh4nks, 2016-02-19:
Hey!
Thanks for doing this!
Is there a way to start ckb minimized? Because just starting the daemon doesn't display the colors on the keyboard (or am I doing something wrong?).
Thanks!

tatokis, 2016-02-19:
It's ckb --background

sh4nks, 2016-02-20:
Ah thanks! Could be quite useful if there was an option in the GUI for this. Should I send a PR with a note on how to run ckb in the background? I haven't found it anywhere..

tatokis, 2016-02-20:
It's up to @ccMSC, but I found it in ~/.config/autostart/ckb.desktop which seems to get added when installing the application.

ccMSC, 2016-02-24:
There was supposed to be a `ckb --help` option which would tell you about this, but it's entirely possible I never got around to adding it.

sh4nks, 2016-02-24:
Haha, no worries! :)
I actually tried to run ``man ckb`` and ``ckb --help`` before opening the issue but none of them where available (no manual entry and no help parameter).
I added the ``ckb.desktop`` file to the directory that @tatokis mentioned and adjusted the start parameter in the .desktop file from ``ckb`` to ``ckb --background``.

Here is the desktop file for those who have the same problem:
```ini
#!/usr/bin/env xdg-open
[Desktop Entry]
Encoding=UTF-8
Version=1.0
Type=Application
Name=ckb
Comment=Corsair keyboard driver user interface
Exec=ckb --background
Icon=ckb
Categories=Settings;System;
```

closed 2016-02-24 by sh4nks