ccMSC/ckb #256: Mac OS 10.11.3, ckb 0.2.4: Reconnecting failures and deamon softlocks

ma-ver-ick, 2016-02-20:
I recently upgraded to 0.2.4 from an earlier 0.2 version (dunno exactly which unfortunately). With the 0.2.4 version I'm unable to reconnect my keyboard once any Corsair keyboard (I have two, one at work and one at home both only connecting to my MBP with ckb). I run the ckb-daemon manually with `sudo ./ckb-daemon`.
The only solution is, to `kill -9 %1` to get the ckb daemon to quit and restart it. It worked with an earlier 0.2.X version without any problems.
I hope that description somehow helps you to figure out the problem. Here is what ckb-daemon is saying:

```
    ckb: Corsair RGB driver beta-v0.2.4
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard  at /var/run/ckb1
[I] Starting input thread for /var/run/ckb1
[E] os_usbrecv (via profile_keyboard.c:28): Got return value 0xe00002d6
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1
[E] os_usbsend (via led_keyboard.c:131): Got return value 0xe0005000
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:38): Got return value 0xe00002c2
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /var/run/ckb1
^C
[I] Caught signal 2
^C
[W] Ignoring signal 2 (already shutting down)
```

The "shutting down" never happens, even after minutes of waiting or unplugging the keyboard.

ccMSC, 2016-02-20:
Sounds like it might be locking up after disconnecting the device. I'll look into it.

ccMSC, 2016-02-28:
Found the issue - there's a pipe block happening whenever a device is removed. I'll have a patch to fix it soon.

ma-ver-ick, 2016-03-22:
Thank you very much! :+1:

ccMSC, 2016-04-08:
Bugfix branch has been merged in v0.2.5. Issue should be resolved.

closed 2016-04-08 by ccMSC

ma-ver-ick, 2016-05-14:
The issue is solved and it works flawlessly again :)

ma-ver-ick, 2016-06-19:
This problem reappeared with the latest beta!

 

>    ckb: Corsair RGB driver beta-v0.2.6
> [I] Root controller ready at /var/run/ckb0
> [I] Connecting Corsair K95 RGB Gaming Keyboard  at /var/run/ckb1
> [E] iterate_devices_hid (usb_mac.c:734): Failed to create device plugin: e00002be
> [E] iterate_devices_hid (usb_mac.c:734): Failed to create device plugin: e00002be
> [I] Starting input thread for /var/run/ckb1
> [E] loadrgb_kb (led_keyboard.c:211): Bad input header
> [W] _start_dev (device.c:45): Unable to load hardware profile
> [I] Setup finished for /var/run/ckb1
> [I] Disconnecting /var/run/ckb1
> ^C^C^C^C^C