ccMSC/ckb #259: [Linux] Using macro to start a graphical program needs the DISPLAY variable to be set first.

penetal, 2016-02-24:
I am on Arch Linux and have the package installed from the AUR, I have enabled the daemon on startup via the systemd unit, and have the ckb program start automatically with gnome login (done through gdm).

When I set a macro key to launch a program it does not inherit any DISPLAY variable so I have to set it before the program launches eg `DISPLAY=:1 gnome-terminal` otherwise it obviously wont show up.

Now I have to set it to `:1` instead of `:0` (or what ever others need) since gdm takes up tty1. Could there maybe be implemented a check-box above the command field that sets this variable automatically?

It is rather cumbersome for new users to know what to set the variable to, or to have a long line like this `DISPLAY=:$(find "/tmp/.X11-unix" -type s -user $(whoami) -o -group $(whoami) | grep -o ".$") gnome-terminal` in the command line box.

ccMSC, 2016-02-24:
Is `libx11` installed? If not, try installing that and then reinstall ckb. ckb should set `DISPLAY` appropriately if it has access to the X11 library, but it's not listed as a requirement so you may have compiled the program without support for it.

penetal, 2016-02-24:
Dosen't seem to be the problem:
```
% pacman -Qi libx11
Name            : libx11
Version         : 1.6.3-1
[...]
Install Date    : Fri 13 Mar 2015 07:40:29 AM CET
[...]
```

ccMSC, 2016-02-28:
Can you think of any reason why `DISPLAY` might not be set for ckb itself? Normally the apps should inherit their environment from ckb, unless it's bugged and the variable is getting deleted (which I'm looking into).

penetal, 2016-02-28:
> Can you think of any reason why `DISPLAY` might not be set for ckb itself?

This seems weird since the ckb program itself shows up as a window just fine.

When I launch it from a terminal that I made sure had `DISPLAY` set before ckb was launched there is still no `DISPLAY` variable when I run a the command `env | sort > hotkey_env` as a hotkey.

Here is a diff of the evironment when ckb is launched from the same terminal window used to create the diff. 
_Commands used_
Hotkey program:  `env | sort > hotkey_env`
Command line ran afterwards: `env | sort | diff - hotkey_env `

Output (username changed to XXX):
```
5d4
< DISPLAY=:1
40d38
< OLDPWD=/home/XXX
43a42
> QT_IM_MODULE=compose
46c45
< SHLVL=1
---
> SHLVL=2
54c53
< WINDOWID=62915506
---
> WINDOWID=62914566

```

ccMSC, 2016-02-28:
There's a function near the bottom of `src/ckb/keyaction.cpp`:
```c++
void KeyAction::adjustDisplay(){
#ifdef USE_LIBX11
    char* display_name = XDisplayName(NULL);
    if(!display_name)
        return;
    Display* display = XOpenDisplay(display_name);
    if(!display)
        return;
    char* display_string = DisplayString(display);
    if(!display_string)
        return;
    size_t envstr_size = strlen(display_string) + 4;
    char* envstr = new char[envstr_size];
    strncpy(envstr, display_string, envstr_size);
    envstr[envstr_size - 1] = 0;

    Window root_window = XRootWindow(display, DefaultScreen(display));
    Window root_window_ret, child_window_ret, window;
    XWindowAttributes attr;
    int root_x, root_y, win_x, win_y;
    unsigned int mask_ret;

    XQueryPointer(display, root_window, &root_window_ret, &child_window_ret, &root_x, &root_y, &win_x, &win_y, &mask_ret);
    if(child_window_ret == (Window)NULL)
        window = root_window_ret;
    else
        window = child_window_ret;
    XGetWindowAttributes(display, window,  &attr);

    char* ptr = strchr(envstr, ':');
    if(ptr){
        ptr = strchr(ptr, '.');
        if(ptr)
            *ptr = '\0';
        char buf[16];
        snprintf(buf, sizeof(buf), ".%i", XScreenNumberOfScreen(attr.screen));
        strncat(envstr, buf, envstr_size - 1 - strlen(envstr));
    }
    setenv("DISPLAY", envstr, 1);

    delete[] envstr;
    XCloseDisplay(display);
#endif
}
```

Could you try erasing the whole thing and recompiling?
```c++
void KeyAction::adjustDisplay(){
#ifdef USE_LIBX11
#endif
}
```

This should confirm that it's a bug in ckb as opposed to an issue with the display manager.

penetal, 2016-02-28:
After patching the file, compiling and running ckb program from there I get the `DISPLAY` variable I should get when I output the environment with a hotkey press. I also tested with `gnome-terminal` and the window comes up as expected without any problems.

Here is the procedure I used after cloning the repo if you wish to know, hostname and username changed obviously:

`XXXXX@YYYYY ~/ckb/ckb/bin (git)-[master] % pkill ckb ` 

`XXXXX@YYYYY ~/ckb/ckb (git)-[master] % git diff`

```
diff --git a/src/ckb/keyaction.cpp b/src/ckb/keyaction.cpp
index 2fef6b6..21f06bb 100644
--- a/src/ckb/keyaction.cpp
+++ b/src/ckb/keyaction.cpp
@@ -424,45 +424,5 @@ void KeyAction::keyEvent(KbBind* bind, bool down){
 
 void KeyAction::adjustDisplay(){
 #ifdef USE_LIBX11
-    char* display_name = XDisplayName(NULL);
-    if(!display_name)
-        return;
-    Display* display = XOpenDisplay(display_name);
-    if(!display)
-        return;
-    char* display_string = DisplayString(display);
-    if(!display_string)
-        return;
-    size_t envstr_size = strlen(display_string) + 4;
-    char* envstr = new char[envstr_size];
-    strncpy(envstr, display_string, envstr_size);
-    envstr[envstr_size - 1] = 0;
-
-    Window root_window = XRootWindow(display, DefaultScreen(display));
-    Window root_window_ret, child_window_ret, window;
-    XWindowAttributes attr;
-    int root_x, root_y, win_x, win_y;
-    unsigned int mask_ret;
-
-    XQueryPointer(display, root_window, &root_window_ret, &child_window_ret, &root_x, &root_y, &win_x, &win_y, &mask_ret);
-    if(child_window_ret == (Window)NULL)
-        window = root_window_ret;
-    else
-        window = child_window_ret;
-    XGetWindowAttributes(display, window,  &attr);
-
-    char* ptr = strchr(envstr, ':');
-    if(ptr){
-        ptr = strchr(ptr, '.');
-        if(ptr)
-            *ptr = '\0';
-        char buf[16];
-        snprintf(buf, sizeof(buf), ".%i", XScreenNumberOfScreen(attr.screen));
-        strncat(envstr, buf, envstr_size - 1 - strlen(envstr));
-    }
-    setenv("DISPLAY", envstr, 1);
-
-    delete[] envstr;
-    XCloseDisplay(display);
 #endif
 }
```

`XXXXX@YYYYY ~/ckb/ckb (git)-[master] % qmake`
`XXXXX@YYYYY ~/ckb/ckb (git)-[master] % make`

`XXXXX@YYYYY ~/ckb/ckb/bin (git)-[master] % ./ckb `
Set G1 key to "env | sort > patch_env"
 ```

XXXXX@YYYYY ~/ckb/ckb/bin (git)-[master] % cat patch_env | grep -i "DISPLAY="
DISPLAY=:1
```

ccMSC, 2016-02-29:
Thanks. Should be fixed properly by 8c18c78264958d7dc23e51ac650f0a38b9a950d1 (`bugfix` branch).

penetal, 2016-02-29:
> Thanks. Should be fixed properly by 8c18c78 (bugfix branch).

Confirmed working on my system. Thank you very much.

closed 2016-04-08 by ccMSC