ccMSC/ckb #260: Please Help.

BrandonGood, 2016-02-25:
So, https://gyazo.com/b867639163e58667c1f1b5d4461058b7
This is what happens when I open the program. I can't change the lighting, I'm not trying to bind keys. I'm trying to change lighting, and there's no option for it. Can someone please help?

ccMSC, 2016-02-28:
Unfortunately, the lighting on the non-RGB K70s isn't programmable by software.

BrandonGood, 2016-02-29:
So I can't change the lighting at all? Thank God it came in red...

ccMSC, 2016-02-29:
There's a button on the keyboard which allows you to program the lights - there should be instructions for doing so in your manual. But it's static lighting only (no animations) and there's no other way to change it.

closed 2016-02-29 by ccMSC