ccMSC/ckb #263: FreeBSD

Bean6754, 2016-03-02:
CKB is unable to compile on FreeBSD due to a make error. It is unable to find targets to compile. I believe it is something to do with the command:  make -j 24 "$@" 2>$TMPFILE

ccMSC, 2016-03-02:
I don't know about the GUI, but the driver won't compile on FreeBSD because it's neither Linux nor OS X. We'd need support for the equivalents of udev, usbfs, and uinput. I'm not sure if it would be possible to create a *BSD wrapper for the Linux code or if those sections would have to be rewritten completely.

Bean6754, 2016-03-02:
Thank you. :)  Could you please find a way to port it over to BSD.

Bean6754, 2016-03-02:
Also, the GUI should work fine, becuase I have Qt4 and Qt5 installed on my FreeBSD system.

Bean6754, 2016-03-02:
Thank you :+1:

ccMSC, 2016-03-03:
I don't know if I'll ever have time for a BSD port, to be honest. I unfortunately can't work on this project very much and the backlog is already long. I'd like to leave this open in case anyone else wants to take it on, though.