ccMSC/ckb #265: Macro keys for Scimitar

itsmillertime65, 2016-03-03:
Not sure this is the correct place to post this, but I'm trying to figure out how to bind macros to specific keys on the Scimitar mouse . . . For instance I'd like for the ⌘ + = to be mapped to my 1 key on the mouse numpad to be able to zoom in. I can't figure this out and I noticed the documentation said advanced macros aren't supported, but does that mean no macros are supported at all?

frickler24, 2016-03-05:
@itsmillertime65 currently the first version of macro handling for keyboards is in the testing stage.
I've just tried what happens when I try to program a mouse key with the macro mechanism, but it fails.
So currently you can only use the Mouse-Binding function "Keyboard, Mouse, Special and Program".
If some more user find it userful to program their mouse-keys, I will look for a solution.

frickler24, 2016-03-05:
@itsmillertime65 a correction: I've tried id with a M65, not a scimitar. So I am not sure what mechanism is used by the scimitar to send macro information.
You can try it by yourself: Define a Keyboard-Macro (hope, you have a keyboard with macro function), copy the resulting macro-definition to the correspondig panel for your mouse. Apply and look what happens.

itsmillertime65, 2016-03-07:
@frickler24 Thanks for the help! I ended up downloading Keyboard Maestro and applying macros to the F keys there. I then bound my mouse buttons to the specific Function key that I have Maestro set up to a macro. For instance, I have the F1 key in Keyboard Maestro set to the ⌘ + = macro and I have my 1 button on the  Scimitar numpad mapped to the F1 key . . . this way whenever I push the 1 key on my mouse Keyboard Maestro tells my Mac to zoom in.

frickler24, 2016-03-07:
@itsmillertime65 ok, going this way IT Workshop, gut you block tho keys - one at the KB, one at the mouse.
Have you tried the following way: 
1. Open the CKB Bind-Macro tab.
2. select f2 key in the UI at the mouse
3. Type "+a,-a" or something similar into the left macro keys textpane and klick Apply.
4. press f2 e.g. in the Macrotext pane 

What Happens?
Could you post some screenshots?

itsmillertime65, 2016-03-08:
@frickler24 ckb doesn't have a macro tab . . . maybe I'm not on the current version? It will only allow me to pick one of the options below instead of allowing me to pick a modifier key plus another keyboard key.
[image]

frickler24, 2016-03-08:
@itsmillertime65 the version with support of macro keys is up to now the testing branch only.
So either you wait until it is released, or you download and install that branch.

itsmillertime65, 2016-03-09:
Oh, that makes sense. Thanks @frickler24. I may wait until it comes out. I have a working solution now, even though it requires me to run an additional program (Keyboard Maestro) to function.

john-2000, 2016-04-28:
firstly apologies if this is a daft question, secondly thanks for all your work on this - without this I would have not bought a corsair.

finally my question:
my strafe keyboard has a function key and by default f5, f7 - f12 are used as media keys. is assigning other key plus FN cominations to macros or additonal short cut keys like "xf86explorer" etc in spec for the current or future versions? 

thanks

itsmillertime65, 2016-04-28:
Hey @john-2000, I'm not sure you posted this in the correct place . . . this thread was a question I had about the Scimitar mouse. You might want to search in the issues tab to see if anyone else has asked this this question, and if they haven't you can create a new issue. That way someone can find your question easier and probably help you out. 

Good luck!

john-2000, 2016-04-28:
thanks @itsmillertime65

itsmillertime65, 2016-04-28:
You're welcome!

Pestlerogue, 2016-05-08:
Does anyone have an eta on that firmware update? I believe it is v2.01 for Scimitar.  I am really hoping to get that number pad working soon.

itsmillertime65, 2016-05-09:
@Pestlerogue, I too am waiting for this update . . . I tried installing the testing branch of CKB and it was a complete failure. For the time being, I ended up setting my keys on my Scimitar to specific Function keys (F1, F2, etc.) in CKB. I then set those Function keys to shortcuts in the native Keyboard>Shortcuts>App Shortcuts menu. Here you can set specific shortcuts for all applications or on a per app basis. You have to make sure you type in the full, case sensitive name of whatever menu item you're trying to bind a key to on your Scimitar. For instance, "Zoom In" is bound to "F1", which is then bound to the "1" key in CKB on my Scimitar.

For now this method is working for me, but if the shortcut you're trying to access doesn't appear in the top menu bar then there's no way to access the shortcut in the Mac Shortcuts menu. Let me know if you need any help setting this up and I can try and help you out by posting a short video or pics.

ccMSC, 2016-05-10:
@Pestlerogue The current Scimitar firmware should be compatible with ckb. Though there are some connection issues on the Mac side that I'm still trying to sort out.

v0.3 is going to have some basic macro support, but it's been getting a lot of delays due to firmware updates bricking the driver, so no ETA yet.

Pestlerogue, 2016-05-11:
@ccMSC @itsmillertime65 My issue is perhaps more simple.  The number pad on the Scimitar doesn't seem to work at all.  For example, the 1 button on the Scimitar does not perform the same function as the 1 on the keyboard, it does nothing.  I assumed this was a firmware thing.

itsmillertime65, 2016-05-12:
@ccMSC, thanks for the update! @Pestlerogue, My number pad on my Scimitar did the same thing when I first started using it on my Mac. It wasn't until I installed CKB that I could actually get the keys to perform any command/keyboard function at all. If your keys on the Scimitar don't work after binding them to specific keyboard keys/commands then you may have a bad mouse. I would suggest trying to map some of the keys on your mouse to their respective numbers (1 to 1, 2 to 2, etc.) and if after clicking one of those buttons it's still not working then I would try connecting to a PC if you have access to one to see if it's just not functioning at all on your Mac.

TeeJayP, 2016-05-17:
@itsmillertime65  Hi Matt.  Could you please explain to me, or show me how to get the macro keys working like you did? It's proving rather frustrating! Thank you mate.

itsmillertime65, 2016-05-17:
@TeeJayP sure thing! I bound all of my Scimitar keys to a Function Key in CKB (1 to F1, 2 to F2, and so on). I then went into the Mac OS X built-in App Shortcuts manager (System Preferences > Keyboard > Shortcuts > App Shortcuts [see attached image]). 
[image]

Press the + button and then select what application you want the shortcut to work in or if you want it for all applications then leave it as is.
[image]

In the "Menu Title:" you have to put the exact name of the menu item as it appears in whatever app you'd like the shortcut to work in. For example, if I want to create a shortcut to select the next tab in Chrome then I would type in the name exactly as it appears in Chrome's menu (if it's not exact then it won't work, this includes case sensitive characters).
[image]

So for this shortcut you would type "Select Next Tab" and then in the "Keyboard Shortcut:" input you can type in your Function key that you bound your Scimitar to in CKB.

If CKB accepted macro inputs then none of this would be necessary, but until then this has been working for me.