ccMSC/ckb #266: ckb not recognizing both devices

ZereoX, 2016-03-04:
After installing ckb, my Scimitar was detected, but as soon as a plugged in my K95 it became the only visible device. Re-installing did not fix the issue.

Any idea as to why this is?

ccMSC, 2016-03-08:
Does it matter what order you plug them in? Also, does the Scimitar turn off entirely (no input, no lights)? You might try putting them on different USB hubs and see if that helps.

ZereoX, 2016-03-08:
Hey @ccMSC,

it would seem regardless of the USB configuration (USB 3.0-3.0, 2.0-2.0, 3.0-2.0) my scimitar is no longer detected by CKB regardless of the order it is plugged in (even when plugged in by itself). I can confirm that both my K95 and Scimitar work in unison under Windows with CUE. 

As to your other question, my scimitar has basic functionality (Left and Right click, mouse wheel + click). All LEDS are turned off except the DPI light (which works correctly).
The K95 can also be unplugged and replugged and will be detected properly by CKB. As for the Scimitar, only a full reboot will allow it to be re-detected by the OS (not ckb).

ccMSC, 2016-03-08:
Is this on a Linux installation? If so, check your cmdline parameters (https://github.com/ccMSC/ckb#troubleshooting)

The correct code for K95 RGB + Scimitar RGB is `usbhid.quirks=0x1B1C:0x1B11:0x20000408,0x1B1C:0x1B1E:0x20000408`

ZereoX, 2016-03-08:
Sorry I forgot to mention that I had already attempted to add them to my kernel cmdline. 

I've attached a copy of my grub.
[grub.txt](https://github.com/ccMSC/ckb/files/162859/grub.txt)

ZereoX, 2016-03-26:
It would seem the [fw2](https://github.com/ccMSC/ckb/tree/fw2) branch with firmware v2.01 [fix](https://github.com/ccMSC/ckb/commit/728d9c0229bba4b1946ef744e215aa80e46bb777) has solved my issue.

closed 2016-03-26 by ZereoX