ccMSC/ckb #267: [Minor Annoyance] Scandinavian keyboard is named Swedish

kimamb, 2016-03-05:
Well, the title describes it. The scandinavian keyboard is named swedish. And as a norwegian I don't like that.

(Even the packaging states "Scandinavian" on the front)

This would be easy to fix, just change a line in the code?

ccMSC, 2016-03-08:
Since the buttons are flipped I'll just make a new layout. I'll try to upload it tomorrow.

kimamb, 2016-03-08:
Norwegian and Swedish are the same (different looks but still exact same function) Danish is the odd one. 

Sorry if you already knew this, I just thought I was that clear.

ccMSC, 2016-03-10:
Added in ca2cb7c035a777a91d6827abe23bb6389520cc08. I put both layouts separately, in addition to the existing Swedish layout, since ckb's layouts are usually separated by country.

kimamb, 2016-03-10:
Thanks. Everything seems to be in order.

Good job, I appreciate it.

closed 2016-04-08 by ccMSC