ccMSC/ckb #271: ckb0.2.4. unplug lose original setting

peteryu0506, 2016-03-14:
I use corsair keyboard for my mac. After I followed the instruction in order to make the keyboard work. I installed the ckb. It works perfectly but the only issue is that every time I unplug the keyboard and the next time i reconnect it, my laptop will not recognize the keyboard and the original setting is gone. I have to reinstall the program and then the setting is back. I'm wondering if there is any way could solve it coz every time it take about 20mb to install.

Thank you so much!

ccMSC, 2016-03-15:
It's being worked on! I'm hoping to post a fix for this in the next 1-2 weeks. In the mean time, if you have any USB hubs around, try using one of those and see if it helps.

peteryu0506, 2016-03-15:
thank you so much!

jspies, 2016-03-31:
Thanks for working on this. I have the same issue, even through a USB hub. I use a macbook air and have to reinstall every time I plug the laptop in.

amsierra, 2016-04-02:
Same here. If I can provide any debugging details please let me know.

jonstonis, 2016-04-04:
I have the same issue.  Any ETA on a published fix?

ccMSC, 2016-04-08:
Should be fixed as of v0.2.5. If you're still having the issue, please join the discussion in #233

closed 2016-04-08 by ccMSC

jspies, 2016-04-13:
Installed today and so far I'm 2 for 2 on plugging back in. Thanks for the fix!

jspies, 2016-04-19:
Update after week:

It's hit or miss. Better than the previous version but about half of the time I have to reinstall the drivers to get my keyboard detected. I'm happy to post in Issue #233 but reading through the thread seems to be a little different.