ccMSC/ckb #279: CKB does not open on El Capitan

EvN-SKO, 2016-03-30:
_NOTE: great job with ckb, works like a charm on my iMac (OSX 10.11.4)_

I am trying to customize my Strafe RGB more, however, CKB won't open when I try clicking it. Switching USB ports, restarting / rebooting, uninstalling and reinstalling the pkg, I've tried almost everything.

If @ccMSC or anyone else can help here, that'd be great! TYSM!

**TL;DR: CKB won't open to let me customize, need solutions (temporary or permanent)**

ccMSC, 2016-03-31:
Same issue on my machine - the tray icon works, but relaunching the app doesn't. Will try to fix soon. Meanwhile, try running this in a Terminal:
```
/Applications/ckb.app/Contents/MacOS/ckb
```

EvN-SKO, 2016-03-31:
TY for fix!

Another quick Q before I close: any possible way to save more than 1 mode / profile on keyboard, or is that a hardware limitation?

ccMSC, 2016-04-01:
Unfortunately no, it's a hardware limit. There's no way to save animations or bindings either.

EvN-SKO, 2016-04-03:
i see; ugh, my mac is slowly driving my Strafe crazy

every time i boot up my iMac, i have to manually force my Mac to recognize my keyboard, then reinstall ckb in order to get customization

ccMSC, 2016-04-08:
GUI launching is fixed in v0.2.5.

closed 2016-04-08 by ccMSC

EvN-SKO, 2016-04-09:
seems that v0.2.5 doesn't recognize my Strafe RGB now, and I have tried many troubleshooting options

if it is in need of new firmware, that will take me a while, since i have no access to a windows PC around

i created a new issue, #285 - this should address it in fuller detail