ccMSC/ckb #280: Linux Quirks Question

r4d1n, 2016-03-31:
First of all, thank you for making and maintaining this software!

I don't really have an issue with the project, but was just wondering if someone could explain to me what the hex code at the end of the usbhid.quirks grub entries mentioned in the readme actually does. I added the line for my Strafe keyboard and another for my mouse (for which I was able to find the other parameters) and it fixed the errors/slow boot I was experiencing, but for the future and since I am always trying to learn more, I would like to better understand what is happening here.

I haven't been able to find a decent high level answer by googling -- it's all either specific fixes like this, or very nitty gritty discussions of Linux C code -- so I'd really appreciate an explanation if anyone has time to write one up.

Thanks again.

ccMSC, 2016-04-01:
The quirks are defined here: https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/include/linux/hid.h?id=refs/tags/v4.4.6#n301

ckb's suggested value of `0x20000408` comes from combining the following three:

`0x00000008` is `HID_QUIRK_NOGET`, which instructs the driver not to read certain status values from the device.

`0x00000400` is `HID_QUIRK_ALWAYS_POLL`. This instructs the driver to accept *all* input packets from the device regardless of whether or not it understands them (by default, packets which are not understood are rejected, which leads to the keyboard hard-locking).

`0x20000000` is `HID_QUIRK_NO_INIT_REPORTS`, which prevents unnecessary initialization packets from being sent.

These were found mostly by experimentation - it seems this particular combination solves the startup issues for most people. Not sure what (if any) effects the other values would have, but I didn't want to add more quirks without a good reason for it.

r4d1n, 2016-04-01:
That's very helpful. Thank you!

closed 2016-04-01 by r4d1n