ccMSC/ckb #281: Do you intend to support corsair headsets?

primetoxinz, 2016-04-01:
I have yet to see anywhere if headsets, such as the VOID, will be supported by this

ccMSC, 2016-04-04:
I do have access to a VOID headset so it's definitely possible. Haven't really looked into it yet due to a lack of demand, but I'll consider it for v0.3.

Serubin, 2016-04-21:
Just to bump. This would be and extremely nice addition.

bqcrabtree, 2016-07-02:
(Sorry for the slight "necro"-bump, but you mentioned lack of demand.)

I would like to request the same. I use a Linux machine as my development machine, and I use an M65 mouse, STRAFE keyboard, and a Void RGB Wireless headset while I work. RGB support for the Void would be AWESOME, so I can get my purple color scheme no matter which machine I'm on.

sattva1, 2016-07-14:
Same here. Even a mere option to disable RGB lighting altogether would be beneficial to save battery power.

orbit42, 2016-08-09:
Ditto, I would also accept a simple method just to turn of the light. I feel like a damn lightbright when I wear it.

bwoodruff, 2016-08-24:
++ would like to see (even limited) VOID support.

Lemmmy, 2016-11-08:
any update on this?

bwoodruff, 2016-11-08:
It appears this project is abandonware. It hasn't been updated even for bug fixes since June. It seems unlikely that we'll see new features unless another developer, or group of developers, steps up and takes the project over.

Lemmmy, 2016-11-08:
rip :(

popcatalin118, 2016-11-16:
Please add corsair void wireless as an option. If you do i will repurchase that headset you're my only hope :(

bwoodruff, 2016-11-16:
@popcatalin118 see my comment here: https://github.com/ccMSC/ckb/issues/281#issuecomment-259024537