ccMSC/ckb #282: Demo Profiles gone??

Artis4n, 2016-04-04:
all the demo profiles went and reinstalling doesn't add them??

ccMSC, 2016-04-08:
The demo profile currently can't be re-added after it's removed. If there's a specific animation you want back, I can give you instructions on how to re-create it.

Artis4n, 2016-04-08:
I liked all three demo ones :(

audreyseo, 2016-04-19:
I'd like instructions specifically for the "trippy" one, if you don't mind!

Also, just wondering, will updating in the future remove the profiles as well?

ccMSC, 2016-04-21:
1. **Rainbow**
 * To start, set all keys to white (255, 255, 255). http://i.imgur.com/BLRrX0Z.png
 * Select all keys except WASD and arrow keys, as shown: http://i.imgur.com/F3xNZ7a.png
 * Click `New Animation`. Select the `Wave` animation with the `Rainbow` preset. https://i.imgur.com/JbLV9Cp.png
 * Change the `Angle` parameter to 120° and the `Duration` to 3.0. Leave all other parameters the same. https://i.imgur.com/6c014vI.png
 * Select *all* keys.
 * Click `New Animation`. Select the `Gradient` animation with the `Type Lighting` preset.
 * Click OK - no need to change parameters on this one.
2. **Breathing**
 * Start by setting all keys to white.
 * Set the G-keys (if you have them) and the NumPad keys to light red (255, 128, 128). http://i.imgur.com/ImvpnA4.png
 * Set the typing keys and the arrow keys to red (255, 0, 0). http://i.imgur.com/nrjYRhy.png
 * With these keys still selected, click `New Animation`. Choose `Gradient` with the `Fade in` preset.
 * Click on the `Color` button to change the gradient.
 * Click near the middle of the gradient bar to create a new point. Set its position to 50%. https://i.imgur.com/sQFzAxH.png
 * Change its color to white (255, 255, 255) and its opacity to 100%. https://i.imgur.com/hyeC1rr.png
 * Click on the leftmost point and change its opacity to 0%. https://i.imgur.com/UCkiN8a.png
 * Click OK to save the gradient.
 * Change the animation's duration to 5.0 seconds. https://i.imgur.com/YEIHW0M.png
 * Click on the `Timing` tab and enable `Repeat: Forever` near the top. https://i.imgur.com/2xbUnkO.png
 * Click OK.
3. **Trippy**
 * Set all keys to this purple color (170, 0, 255). http://i.imgur.com/4X91KRm.png
 * With the keys still selected, click `New Animation`. Select the `Random` animation with the `Default` preset.
 * Change the `Duration` to 0.5 seconds. Click OK. https://i.imgur.com/47ChKYq.png
 * Near the bottom of the lighting screen, change the `Blend mode` to `Multiply`. https://i.imgur.com/pcP8YbN.png
 * Select all keys again. Create a new animation, using `Ripple` with the `Type Lighting` preset.
 * Change the `Ring Length` to 10%. https://i.imgur.com/Sv4j1Zy.png
 * Open the gradient editor and change the first point (selected by default) to magenta (255, 0, 127).
 * Click OK and exit out of the animation properties. https://i.imgur.com/C0FfB6z.png
 * Change the animation's `Blend mode` to `Add`. https://i.imgur.com/1P15rxJ.png

Sorry it's so complicated - I'll try to add a way to get the profile back in a future version.

closed 2016-06-13 by ccMSC