ccMSC/ckb #284: [OSX] High CPU usage without connected keyboard

flindeberg, 2016-04-06:
Using v0.24-beta the ckb-process uses a non-significant amount of CPU-time. In my case writing in emacs and compiling with pdflatex every now and then together had less CPU-time than ckb.

My solution was simply stopping the process, but it is annoying to start and stop it every time i "dock" my MBP. 

Is this something known?

frickler24, 2016-04-06:
@flindeberg I checked it at my  Linux Mint 17.3 with I7-4771CPU.
Keyoard active: ckb client needs 2.3 - 3.3% CPU
Keyboard unplugged: ckb needs 0.3% max.

flindeberg, 2016-04-06:
@frickler24 I'm using OSX 10.11.4 and it goes between roughly 0.2% and 12%, often fluctuating up towards 12%. 

I'll double check the values for connected CPU usage next time I'm at the office. I just noticed it today out of office (when CPU usage and battery time actually matters).

To update on the CPU-time with some more data, during a reading / writing session (ie mendeley + emacs + biber + pdflatex etc) lasting roughly 1h30m ckb had more CPU-time than all other processes _combined_. Compile time for that skeleton of a paper is mere seconds though, so not CPU intensive work.

If it matters somehow I've installed ckb via homebrew, ie `brew cask install ckb`, but the md5 seems to match the one on the "latest releases" page.

flindeberg, 2016-04-07:
With keyboard it is between 2% and 6%. It seems a lot, is it supposed to be that high?

I haven't had time to build it myself and see what is using the CPU that much, but I'll see if I can get enough time this weekend.

CMasterson, 2016-04-07:
The app cbk-deamon was heating my Macbook up so much that my fans were constantly spinning up and the laptop temps were constantly above 60 degrees celsius.

Had to uninstall.

eohernandez, 2016-04-07:
Have the same problem, every time I unplug the keyboard, both processes (ckb  and ckb-daemon) use between 30% to 50% of CPU. The only way to stop is killing the processes, but if you plug in and unplug the keyboard you need to kill both processes again.

ccMSC, 2016-04-08:
Should be fixed in the new v0.2.5. Previous version sometimes had an infinite loop caused by unplugging a device.

flindeberg, 2016-04-08:
@ccMSC Awesome, much better now.

It seems to stay around 0.5% but sometimes still jumps to 12-15%. Do you know why it jumps every now and then?

flindeberg, 2016-04-11:
@CMasterson Sometimes I have the issues you are describing as well, ie cbk-daemon chugging a significant CPU-load, approx 50%, ie one core for me. Usually in conjunction with connecting / disconnecting and "bus errors". In 0.2.4 it was actually the cbk process which took cpu in the 2%-12% interval.

ccMSC, 2016-04-13:
CPU usage jumps sometimes when there's a lot of keyboard activity. I've made efforts to reduce it, but at this point there's not much I can do without compromising functionality.

securify, 2016-04-15:
Same issue here. El Capitan on a new retina 13inch macbook pro. CPU usage seems to jump while connected with spikes up to 45% on single core and when disconnected it fires more often and is consistent about 15%.

closed 2016-06-13 by ccMSC