ccMSC/ckb #286: Linux Mint not detecting keyboard after sleep (Suspend)

fulldesign, 2016-04-10:
Awesome driver! Just having issues with Linux Mint, where if I Suspend/ Sleep my PC. When I wake it up again (able to wake the computer with any key tap) but cannot enter log in password as the keys aren't recognised. This is also a problem on start-up as I have to set it to autologin to bypass the log in screen as again the keys aren't recognised until I log in and CKB auto starts. 

Interestingly on the login screen if I hit something like Ctrl, the password box will appear, suggesting the keyboard is recognised, just doesn't seem to be able to interpret it. I'm guessing this is something to do with it being a user run process. It there any way to get the daemon running before login? At the minute I need to leave the computer running and just lock the screen which is less than ideal.

Thanks

SlowLogicBoy, 2016-04-11:
I have similar problem:
If I connect my PC to outlet and wait a bit my keyboard turns into demo mode and if I turn on my pc Linux Mint login screen doesn't recognize my keyboard (I use on screen keyboard then), but if I turn on my pc just after I put it into outlet my keyboard is detected normaly. I think these problems are related.

fulldesign, 2016-04-11:
Ah, completely forgot about the screen keyboard! A good temporary solution for sure. Beats having to plug in my spare crappy keyboard. I wish Corsair would make the keyboards integrate as normal keyboards without all the flashy LED patterns needing special drivers just so we can log in first!

ccMSC, 2016-04-13:
AFAIK it's a hardware issue and there's no way I can resolve it. Sorry about that.

fulldesign, 2016-04-14:
Ah yeah figured it might be out of your control. Thanks anyway!

closed 2016-04-14 by fulldesign

SlowLogicBoy, 2016-04-14:
It might be, because when I unplug both keyboard usb ports and plug them back in, the keyboard works normaly.

tatokis, 2016-04-14:
Since unplugging and plugging the keyboard back in fixes it,
wouldn't doing an ioctl() with USBDEVFS_RESET also have the same effect?