ccMSC/ckb #289: [Linux] CKB prevents Logout/Shutdown

FiraWolf, 2016-04-11:
When ckb (the program, not the daemon!!) is running, a logout or shutdown via User Interface is prevented almost every time. I have to at least press logout 2 times or it's not even working at all.
Using "sudo shutdown now" works right off the bat, though.

systemd/journal is reporting:
`ksmserver: Client  "/usr/bin/ckb"  ( 10100155ccd1000146040930700000009100006 ) canceled shutdown.`

With ckb manually shut down before logout, it's working as expected.

I'm using Antergos with 4.4.5-1-ARCH Kernel and the latest KDE Plasma DE (5.6.2) and a K70-RGB with enabled kernel boot parameters.

ccMSC, 2016-04-13:
Is there any way to make it say *why* the shutdown was canceled? Typically it just sends a close signal to the app which causes it to exit. I'm assuming that doesn't happen here for some reason.

FiraWolf, 2016-04-13:
Unfortunately, there isn't very much information given. But maybe I just don't know in which log to look. I was checking logs for days and found some new files every day. They are just too distributed so that I probably overlooked an important one.

frickler24, 2016-04-19:
What happens if you terminate ckb client manually before doing shutdown? It's just to verify that it is the client, not the daemon cancelling the shutdown request.

FiraWolf, 2016-04-19:
If I terminate ckb client manually (making sure ckb daemon is still running) the shut down process is not interrupted.

However, there was an KDE update and it seems it fixed the problem. I'll further investigate and let you know if the problem occurs again. For now don't mind it.

closed 2016-06-13 by ccMSC

zicodxx, 2016-07-20:
I recently bought a Strafe RGB and use it with ckb on Arch Linux (thank you for this great program - it's because of you, that I bought this keyboard in the first place).
However I am running into the same issue (also related to KDE (16.04) and Plasma (5.7.1)):
> ksmserver[10319]: ksmserver: Client  "/usr/bin/ckb"  ( 1078656e6f000146901928800000103190010 ) canceled shutdown.
If I just kill or quit /usr/bin/ckb prirot to shutdown, it works perfectly. 

Is there any other info I could provide for you to solve this issue?