ccMSC/ckb #290: Run Macro/G key program in default shell

penetal, 2016-04-12:
Under _Binding_ of keys to _Program_ it would probably be nice for less experienced users to have a simple checkbox next to the command window to run said command in the user's default shell (cmd on windows I would assume).

The reason I am asking for this is because as it is now we have to manually write it in as something like
`sh -c 'pgrep -f "^mono.*keepass" && wmctrl -a keepass || keepass'`
to be able to run something that is just starting an executable, not a big deal, but it would be a nicety to have.

ccMSC, 2016-04-13:
It does, actually. `pgrep -f "^mono.*keepass" && wmctrl -a keepass || keepass` should be equivalent to what you wrote, unless it's getting a different `sh` for some reason?

penetal, 2016-04-13:
Sorry, I was just being stupid, it was [#259](https://github.com/ccMSC/ckb/issues/259) showing itself again because I weren't thinking. Switched the line to 
`pgrep -f "^mono.*keepass" && DISPLAY=:1 wmctrl -a keepass || keepass` instead of
`DISPLAY=:1 pgrep -f "^mono.*keepass" && wmctrl -a keepass || keepass` and it worked.

Didn't include the variable because I know you fixed it in a newer version. I am using the build from AUR so I'm still on beta-v0.2.1 so the fix for the `DISPLAY` variable isn't yet active.

I'm just gonna close this.

closed 2016-04-13 by penetal