ccMSC/ckb #291: Can't compile testing branch: Q_DECL_OVERRIDE in macroreader.h

fleischie, 2016-04-13:
Hey,

in `macroreader.h` the statement:
```
void run Q_DECL_OVERRIDE ();
```
prevents compilation on my machine (Arch Linux with gcc 5.3.0)

Writing the above statement as:
```
void run () Q_DECL_OVERRIDE;
```
works as expected.

As I cannot imagine, that this issue wasn't apparent before, I wanted to make sure that this is a legit issue. Furthermore I wanted to ask, whether this would make for a good pull request? (Because I wanted to make sure, this is actually breaking for somebody else, too. :) )

ccMSC, 2016-04-18:
Same issue here...I haven't actually compiled the testing branch in over a month. Added a commit to fix it.

frickler24, 2016-04-19:
Thank you both for changing the declaration.
In my env (gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.1)) both versions are compiled w/o warning or error...

fleischie, 2016-04-25:
Seems fixed. :)

closed 2016-04-25 by fleischie