ccMSC/ckb #295: M65 RGB detected, but won't actually move the cursor when moved

predme, 2016-04-14:
It worked flawlessly on my last install but I recently reinstalled Arch and now, no longer works. The keyboard works just fine, and according to ckb-daemon both devices are detected and connected. Opening ckb allows me to change the lighting effects of the mouse and various other settings (lighting also changes on the mouse), but moving it? That does not work. 

Any idea whats up? I've added both the usbhid.quirks to the startup line which doesn't help. All it does is speed up booting but doesn't fix the mouse problem.

ccMSC, 2016-04-18:
If you use `sudo systemctl stop ckb-daemon` *without* removing the mouse first, does it start working again? I'd also try turning the DPI up or down just to see if it does anything.

predme, 2016-04-18:
I stopped the daemon with the mouse connected and still no response. I let it sit for a few minutes, tried again, then unplugged and replugged and still no life. Changing the DPI seems to have no effect either. Forgot to mention that I have a K95 RGB as well and there are no issues with this keyboard. It works with both the daemon on and off, but this mice is a little more stubborn. 

Starting ckb-daemon back up manually just outputs that it connects both the devices, starts the input threads and that the setup finishes. No errors. The daemon clearly knows that both devices exist as I can still configure them using the ckb software (lighting etc) so it's odd that I can't actually move the mouse around.

closed 2016-04-18 by predme