ccMSC/ckb #296: ckb segfaults on Ubuntu 16.04

rtrevor, 2016-04-15:
Only recently got a Corsair K70 RGB and installed ckb on an install of Ubuntu 15.10, worked fine and had no issues.

Since upgrading to 16.04 ckb keeps segfaulting. I've tried re-installing ckb so that it compiles against the latest GCC / libraries but it hasn't made any difference.

Haven't noticed any way to reliably repeat it, it just crashes out every few hours.

Terminal output is:

$ ckb
Scanning  "/usr/bin/ckb-animations/ckb-gradient"
Scanning  "/usr/bin/ckb-animations/ckb-pinwheel"
Scanning  "/usr/bin/ckb-animations/ckb-rain"
Scanning  "/usr/bin/ckb-animations/ckb-random"
Scanning  "/usr/bin/ckb-animations/ckb-ripple"
Scanning  "/usr/bin/ckb-animations/ckb-wave"
Downloaded new firmware list. 6 entries found.
ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.
Segmentation fault (core dumped)
$

Any ideas / how can I provide more info to help with this?

tatokis, 2016-04-15:
Please try resetting it, by removing its config files. 
Additionally, run ckb with gdb, and once it segfaults, post the output of "bt full"

rtrevor, 2016-04-27:
Hi

I managed to get a backtrace after starting ckb with a clean config, here is the full output:

(gdb) run
Starting program: /home/xxx/Downloads/ckbdebug/ckb-master/bin/ckb 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe888e700 (LWP 27768)]
[New Thread 0x7fffe15dc700 (LWP 27769)]
[New Thread 0x7fffe0ddb700 (LWP 27770)]
[New Thread 0x7fffdbfff700 (LWP 27771)]
[New Thread 0x7fffdb390700 (LWP 27773)]
[New Thread 0x7fffdab8f700 (LWP 27774)]
[New Thread 0x7fffd9edc700 (LWP 27775)]
[New Thread 0x7fffc4b96700 (LWP 27776)]
[New Thread 0x7fffbf7bf700 (LWP 27778)]
[New Thread 0x7fffbefbe700 (LWP 27779)]
[New Thread 0x7fffbe0f9700 (LWP 27780)]
[New Thread 0x7fffbd8f8700 (LWP 27781)]
Downloaded new firmware list. 6 entries found.
[Thread 0x7fffdb390700 (LWP 27773) exited]
[Thread 0x7fffdab8f700 (LWP 27774) exited]
[Thread 0x7fffbd8f8700 (LWP 27781) exited]
[New Thread 0x7fffbd8f8700 (LWP 4749)]
[Thread 0x7fffbd8f8700 (LWP 4749) exited]
ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.

Thread 1 "ckb" received signal SIGSEGV, Segmentation fault.
0x00007ffff5fed371 in QSocketNotifier::setEnabled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt full
#0  0x00007ffff5fed371 in QSocketNotifier::setEnabled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#1  0x00007ffff5ee7c8b in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#2  0x00007ffff5f47915 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#3  0x00007ffff5ee255f in QProcess::waitForFinished(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#4  0x00007ffff5ee2d7e in QProcess::~QProcess() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#5  0x00007ffff4edefe8 in __run_exit_handlers (status=1, listp=0x7ffff52685f8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true) at exit.c:82
        atfct = <optimised out>
        onfct = <optimised out>
        cxafct = <optimised out>
#6  0x00007ffff4edf035 in __GI_exit (status=<optimised out>) at exit.c:104
No locals.
#7  0x00007ffff41ecea1 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
No symbol table info available.
#8  0x00007ffff78c2cee in _XIOError () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#9  0x00007ffff78c057d in _XEventsQueued () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#10 0x00007ffff78b1f47 in XPending () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#11 0x00007ffff41e2e2e in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
No symbol table info available.
#12 0x00007ffff3b3b8ad in g_main_context_prepare () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#13 0x00007ffff3b3c24b in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#14 0x00007ffff3b3c42c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#15 0x00007ffff6008a7f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#16 0x00007ffff5fafdea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#17 0x00007ffff5fb7e8c in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#18 0x000000000041f15c in main (argc=1, argv=0x7fffffffdf48) at main.cpp:124
        a = <incomplete type>
        background = false
        w = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff6dc8e80 <QMainWindow::staticMetaObject>, stringdata = 0x527920 <qt_meta_stringdata_MainWindow>, data = 0x527c40 <qt_meta_data_MainWindow>, 
              static_metacall = 0x4ec220 <MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, static mainWindow = 0x7fffffffdd90, settingsWidget = 0x9e7440, 
          kbWidgets = {<QListSpecialMethods<KbWidget*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = -1}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0xc84ff0}, d = 0xc84ff0}}, restoreAction = 0x9ca890, 
          closeAction = 0x9ca8b0, unityDesktop = true, indicator = 0x928d00, indicatorMenu = 0x9148e0, indicatorMenuQuitItem = 0x907e60, indicatorMenuRestoreItem = 0x907ba0, trayIconMenu = 0x10000ffff, trayIcon = 0x0, ui = 0x9be330}
(gdb)

ccMSC, 2016-04-28:
Hmm. Based on the backtrace it sounds like this would be a bug in Qt rather than ckb itself, but it might be something else causing it. I'll see if there's anything wrong with the QProcess usage, though.

tatokis, 2016-04-28:
Since ckb now works with 2.04 on the K70RGB and I am running Ubuntu 16.04, I will keep my eye out and see if I can get more info, as it hasn't crashed for me yet.

closed 2016-06-13 by ccMSC

kaibacorp, 2016-07-22:
I just upgraded from 15.10 to 16.04 and I am having this segfault issue. On Strafe with firmware version 2.04 and ckb beta-0.2.6. So far I've tried resinstalling (with quickinstall) and removing config files in ~/.config/ckb but it still occurs. 

gdb output with clean config:

```
(gdb) run
Starting program: /usr/bin/ckb 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8876700 (LWP 22344)]
[New Thread 0x7fffdd3c5700 (LWP 22375)]
Scanning  "/usr/bin/ckb-animations/ckb-gradient"
Scanning  "/usr/bin/ckb-animations/ckb-pinwheel"
Scanning  "/usr/bin/ckb-animations/ckb-rain"
Scanning  "/usr/bin/ckb-animations/ckb-random"
Scanning  "/usr/bin/ckb-animations/ckb-ripple"
Scanning  "/usr/bin/ckb-animations/ckb-wave"
[New Thread 0x7fffd71a1700 (LWP 22386)]
[New Thread 0x7fffd69a0700 (LWP 22387)]
Starting  "/usr/bin/ckb-animations/ckb-wave"
Starting  "/usr/bin/ckb-animations/ckb-gradient"
[New Thread 0x7fffd5a6d700 (LWP 22420)]
[New Thread 0x7fffd526c700 (LWP 22421)]
Downloaded new firmware list. 6 entries found.
[Thread 0x7fffd526c700 (LWP 22421) exited]
[New Thread 0x7fffd526c700 (LWP 16019)]
[Thread 0x7fffd526c700 (LWP 16019) exited]
ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.

Thread 1 "ckb" received signal SIGSEGV, Segmentation fault.
0x00007ffff5fed371 in QSocketNotifier::setEnabled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt full
#0  0x00007ffff5fed371 in QSocketNotifier::setEnabled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#1  0x00007ffff5ee7c8b in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#2  0x00007ffff5f47915 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#3  0x00007ffff5ee255f in QProcess::waitForFinished(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#4  0x00007ffff5ee2d7e in QProcess::~QProcess() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#5  0x00007ffff4edefe8 in __run_exit_handlers (status=1, listp=0x7ffff52685f8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true)
    at exit.c:82
        atfct = <optimized out>
        onfct = <optimized out>
        cxafct = <optimized out>
#6  0x00007ffff4edf035 in __GI_exit (status=<optimized out>) at exit.c:104
No locals.
#7  0x00007ffff41ecea1 in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
No symbol table info available.
#8  0x00007ffff78c2cee in _XIOError () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#9  0x00007ffff78c057d in _XEventsQueued () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#10 0x00007ffff78b1f47 in XPending () from /usr/lib/x86_64-linux-gnu/libX11.so.6
No symbol table info available.
#11 0x00007ffff41e2e2e in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
No symbol table info available.
#12 0x00007ffff3b3b92d in g_main_context_prepare () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#13 0x00007ffff3b3c2cb in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#14 0x00007ffff3b3c4ac in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#15 0x00007ffff6008a7f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#16 0x00007ffff5fafdea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#17 0x00007ffff5fb7e8c in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#18 0x000000000041e656 in main ()
No symbol table info available.
```

Please let me know if there is further information I can provide!