ccMSC/ckb #297: No compilation.

noah1786, 2016-04-17:
I posted this on the forum but I don't know which would make it to you sooner.

here is a dump of the full error.

http://pastebin.com/kkL5J04h

using Ubuntu 14.04.3 "Trusty" LTS.

Halp.

tatokis, 2016-04-17:
Are you compiling using the ./quickinstall script?
What is your gcc and qmake version?
```
$ gcc --version && qmake --version
```

tatokis, 2016-04-17:
I would like to add that I managed to compile ckb under a relatively clean setup on Ubuntu 14.04.4 LTS having the following dependencies installed.
```
build-essential qt5-qmake libudev-dev qtbase5-dev qtbase5-dev-tools
```
Could you also provide the output of
```
$ apt-cache policy qtbase5-dev
```

noah1786, 2016-04-17:
Here you go!
http://pastebin.com/YRfzmJVp

noah1786, 2016-04-17:
I have those dependencies by the way.

tatokis, 2016-04-17:
It appears that something is wrong with your Qt setup, since the script attempts to compile ckb using both Qt5 and Qt4 headers. (Line 379 in your Pastebin.)

Does the file "/usr/include/qt5/QtCore/QMetaEnum" exist?
If it doesn't, you probably need to purge and reinstall the qtbase5-dev package.

If it does exist, open the file ckb.pro in a text editor, and at the very top, on a new line, add
```
INCLUDEPATH += /usr/include/qt5/QtGui /usr/include/qt5/QtCore
```
Save it, and then run ./quickinstall

If none of this worked, please provide the full output of the quickinstall script (with the gcc/g++ lines).

noah1786, 2016-04-18:
The file did exist, so I put the line you asked me to put in the file. No luck.
I also tried purging and reinstalling qtbase5-dev too . Nothing.
Here is the full output. http://pastebin.com/af1EHJjQ

noah1786, 2016-04-18:
Whoop. Turns out I don't have libappindicator. I'll try installing it from source as my package manager can't find it. Wish me luck.

tatokis, 2016-04-18:
Yeah, I was about to recommend you to try uninstalling qt4-qmake, although I did have it installed on my machine and it worked just fine.

Unfortunately I am not a Qt expert, so I am sure there is a much better way of doing this. Maybe something is wrong with qtchooser.

About libappindicator, libappindicator3-dev should be the package you need
( http://packages.ubuntu.com/search?searchon=contents&keywords=app-indicator.h&mode=exactfilename&suite=trusty&arch=any )

noah1786, 2016-04-18:
I got all packages starting with libappindicator (I had one, I guess the wrong one :/) and it didn't work at first but after a restart everything was do-able. Thanks a ton, really.

closed 2016-04-18 by noah1786