ccMSC/ckb #298: Update - now ckb doesn't recognize my keyboard

athyman, 2016-04-20:
Hello - I just updated my Command Line Tools (OS X 10.11) for Xcode, and when my computer rebooted, the ckb utility no longer recognizes my keyboard. I am able to type on it, but the key bindings I set and the awesome animation I was using no longer work. Is there anything you can do to help? Thank you!

MaxKotlan, 2016-04-20:
I don't know if this relates, but did you update your firmware for the keyboard? I updated to the latest (2.04) and now my keyboard isn't being recognized on OS X as well.

JackFarrand, 2016-04-20:
SAME HERE, USING OSK

JackFarrand, 2016-04-20:
HALP

lsbeuromix, 2016-04-21:
Same issue on Linux build. Ckb shows the keyboard appearing and disappearing every 30 seconds or so. Also on latest firmware (2.04)

ccMSC, 2016-04-21:
Boo - corsair's firmware broke it again :(

I'll try to update as soon as possible.

JackFarrand, 2016-04-21:
thanks man!

athyman, 2016-04-21:
No worries, thank you so much! Love the work you do

philbyjohn, 2016-04-21:
sigh! Should have taken a look here before updating.
Target: K70 RGB
Now usbhid.quirks=0x1B1C:0x1B13:0x4 don't work and dmesg shows
**usbhid 9-1.1:1.2: couldn't find an input interrupt endpoint.** which indicates an enumeration failure.
Very eager to know how you will be fixing this one.

tatokis, 2016-04-21:
I upgraded my K70 RGB to the new firmware in order to see if I can fix the usbhid warning.
Even setting 0x00000004 (HID_QUIRK_IGNORE) didn't get rid of that message, so I don't think there is anything that can be done in order to fix this error without actually editing kernel code.

Ckb doesn't seem to recognise the keyboard at all, however it to works fine with the standard HID driver.

philbyjohn, 2016-04-21:
> however it to works fine with the standard HID driver.

@tatokis Which HID driver is that? I would like to enable it, compile a custom kernel and give it a shot. Please state the CONFIG_ to enable. Are you talking of CONFIG_HID_CORSAIR ?

tatokis, 2016-04-21:
@philbyjohn 
```
tatokis@tatokis-PC:~$ uname -a
Linux tatokis-PC 4.6.0-rc1-customatapatch #1 SMP Sun Mar 27 17:59:35 EEST 2016 x86_64 x86_64 x86_64 GNU/Linux
```
Using the stock Ubuntu kernel config with some unrelated changes.
Could give you the config file if you really need it.

Keep in mind that if you are running with 0x4,the linux HID driver ignores the keyboard completely. You should use the quirk suggested in the readme instead (0x20000408).

It appears that the keyboard does a reset before it "stays on", as the disconnect was not caused by me.
```
[327113.567737] usb 8-2: new full-speed USB device number 9 using xhci_hcd
[327113.724635] usb 8-2: New USB device found, idVendor=1b1c, idProduct=1b13
[327113.724640] usb 8-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[327113.724642] usb 8-2: Product: Corsair K70 RGB Gaming Keyboard 
[327113.724645] usb 8-2: Manufacturer: Corsair
[327113.724647] usb 8-2: SerialNumber: 
[327113.737017] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.0/0003:1B1C:1B13.008E/input/input156
[327113.792035] hid-generic 0003:1B1C:1B13.008E: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input0
[327113.802120] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.1/0003:1B1C:1B13.008F/input/input157
[327113.856196] hid-generic 0003:1B1C:1B13.008F: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input1
[327113.856393] usbhid 8-2:1.2: couldn't find an input interrupt endpoint
[327114.175963] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.0/0003:1B1C:1B13.0090/input/input158
[327114.232045] hid-generic 0003:1B1C:1B13.0090: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input0
[327114.241478] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.1/0003:1B1C:1B13.0091/input/input159
[327114.296099] hid-generic 0003:1B1C:1B13.0091: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input1
[327115.791384] usb 8-2: USB disconnect, device number 9
[327116.203892] usb 8-2: new full-speed USB device number 10 using xhci_hcd
[327116.365647] usb 8-2: New USB device found, idVendor=1b1c, idProduct=1b13
[327116.365650] usb 8-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[327116.365652] usb 8-2: Product: Corsair K70 RGB Gaming Keyboard 
[327116.365653] usb 8-2: Manufacturer: Corsair
[327116.365655] usb 8-2: SerialNumber: 
[327116.378779] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.0/0003:1B1C:1B13.0092/input/input160
[327116.432960] hid-generic 0003:1B1C:1B13.0092: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input0
[327116.442717] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.1/0003:1B1C:1B13.0093/input/input161
[327116.500251] hid-generic 0003:1B1C:1B13.0093: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input1
[327116.500370] usbhid 8-2:1.2: couldn't find an input interrupt endpoint
[327116.784147] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.0/0003:1B1C:1B13.0094/input/input162
[327116.844475] hid-generic 0003:1B1C:1B13.0094: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input0
[327116.854985] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.1/0003:1B1C:1B13.0095/input/input163
[327116.908325] hid-generic 0003:1B1C:1B13.0095: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input1
[327244.672346] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.0/0003:1B1C:1B13.0096/input/input164
[327244.728561] hid-generic 0003:1B1C:1B13.0096: input,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input0
[327244.738586] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:05.0/0000:03:00.0/usb8/8-2/8-2:1.1/0003:1B1C:1B13.0097/input/input165
[327244.792278] hid-generic 0003:1B1C:1B13.0097: input,hiddev0,hidraw5: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:03:00.0-2/input1
```

Some additional info from the ckb daemon:
```
tatokis@tatokis-PC:~$ sudo ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.4
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[E] os_setupusb (usb_linux.c:286): Failed to claim interface: Invalid argument
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
^C
[I] Caught signal 2
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
```

JackFarrand, 2016-04-21:
@tatokis @philbyjohn @lsbeuromix https://www.dropbox.com/s/cdy23fcpsi9oxws/K70RGB_app115.bin?dl=0

The above link is for K70RBG firmware 1.15.  force downgrade using CUE on a windows box like I did and CKB will be working just fine.

@ccMSC let us know if you find anything out about 2.04, from what I can see from the above comments it could break CBK altogether.

philbyjohn, 2016-04-21:
@tatokis I switched back to usbhid.quirks=0x1B1C:0x1B13:0x20000408 and then did a modprobe hid_corsair but that did not work. I think the kernel tries to load hid_corsair at boot but fails. I get 
**usbhid 3-2:1.0: can't add hid device: -110
usbhid: probe of 3-2:1.0 failed with error -110**

I don't see this error in your log message, which might be the reason why it works. Probably because I use the older stock Fedora 22 (64 bit) running 4.4.6-201.fc22.x86_64.

@JackFarrand Did not know we could downgrade. Will go with this workaround in the meantime. Thanks!

JackFarrand, 2016-04-21:
@philbyjohn welcome dude

Serubin, 2016-04-21:
@JackFarrand Was 1.15 the latest version before 2.02?

tatokis, 2016-04-21:
@Serubin 1.33 was the last one. Have a look at https://github.com/ccMSC/ckb/blob/master/FIRMWARE

Serubin, 2016-04-21:
@tatokis Ah thanks. I'm assuming the downloads for that version are gone now?

tatokis, 2016-04-21:
@Serubin searching for "K70RGB_app133_RC.bin" on google seems to not return any results, so I assume so. The links on the FIRMWARE file just point to the new versions.

tatokis, 2016-04-21:
@JackFarrand as I said earlier, I use hid-generic with usbhid, no special driver:
```
tatokis@tatokis-PC:~$ lsmod | grep -i hid
usbhid                 49152  0
mac_hid                16384  0
hid_generic            16384  0
hid                   122880  2 hid_generic,usbhid
```

JackFarrand, 2016-04-21:
@Serubin yep, latest I could find.  seems to work fine, no issues with CKB or CUE. 

On a side note, I only tried CUE recently, does anybody else think that CKB is _FAR_ superior?  The user interface is just so much easier to use.  I think perhaps CKB is less powerful [no scripting engine yet], but CUE makes it really hard to set up a profile properly!

tatokis, 2016-04-21:
@JackFarrand to add to my previous message, when I do # rmmod usbhid the keyboard along with every other HID device dies.
At that point I usually hotplug my PS/2 keyboard and run # modprobe  -v usbhid "quirks=0x1b1c:0x1b13:0x20000000" which reenumerates the keyboard and it starts working again.
(0x20000408 is the same, but I don't need 408 as I do not have the K95's extra keys. It shouldn't affect anything, I just never changed it to the recommended one.)

JackFarrand, 2016-04-21:
@tatokis I think you meant @philbyjohn!

No HID issues here!

tatokis, 2016-04-21:
@JackFarrand Apologies, yes. I lost track of who said what at this point :p

Serubin, 2016-04-21:
@JackFarrand Absolutely, yes. CKB is much better and easier to use.

tatokis, 2016-04-21:
I have only used CUE two times, just to upgrade the firmware. Never set up any profiles on it. CUE isn't even installed on my main computer.
This keyboard was bought just because of CKB, as I rarely use Windows. That alone makes CKB superior for me.

Corsair should acknowledge that at some point, and offer some form of support to @ccMSC for all the great work.

ccMSC, 2016-04-25:
Well this is awkward...it seems only the K70 got the update, not the K95. So I can't actually test it until if/when the K95 version is released.

That said, I have a pretty good idea of what happened. I pushed an update here for Linux users: https://github.com/ccMSC/ckb/tree/fw2kb

Let me know whether or not it's working, and if not, please post a driver log.

For OSX users, open a terminal and run the following commands:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```
I'll need the output of this so I can see where the problem is.

jbickers, 2016-04-25:
The fw2kb version, in Ubuntu 16.04, appears to get the same end result, in that the ckb-daemon runs, but as soon as I run ckb then the keyboard stops responding until I kill the daemon.

I ran the daemon from the command line, and the output was as follows. The last five lines, from "Connecting 3 endpoints" onwards, repeat continuously after that. In this case the keyboard stopped responding immediately, without me having to run ckb.

```
    ckb: Corsair RGB driver beta-v0.2.5+f02
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
```

athyman, 2016-04-25:
Hi - after entering the second sudo command, this prompt appears. However,
my old color scheme and keybindings have returned, so thank you so much!

ckb: Corsair RGB driver beta-v0.2.2

[I] Root controller ready at /var/run/ckb0

[W] iterate_devices (usb_mac.c:378): seize_wait failed, connecting anyway...

[I] Connecting  (S/N: 15024011AEA7A44454E1DEDBF5001940)

[W] iterate_devices (usb_mac.c:378): seize_wait failed, connecting anyway...

[W] iterate_devices (usb_mac.c:378): seize_wait failed, connecting anyway...

[W] iterate_devices (usb_mac.c:378): seize_wait failed, connecting anyway...

[I] Starting input thread for /var/run/ckb1

[E] os_usbrecv (via firmware.c:15): Got return value 0xe00002d6

[W] _start_dev (device.c:24): Unable to load firmware version/poll rate

[I] corsair strafe ready at /var/run/ckb1

sirjaren, 2016-04-25:
Confirming what @jbickers has (Linux 4.5.2, Corsair K70 RGB, ckb HEAD/fw2kb branch):

<pre>
    ckb: Corsair RGB driver beta-v0.2.5+f02
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2

<b>NOTE: This repeats until `ckb` is killed and the `ckb-daemon`</b>

[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
^C
[I] Caught signal 2
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
</pre>

nkm, 2016-04-25:
K70 RGB, Linux 4.4.7, ckb beta-v0.2.5+f02
```
    ckb: Corsair RGB driver beta-v0.2.5+f02
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[I] Attempting reset...
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2

(... same 6 lines repeating ...)

[E] os_usbsend (via device_keyboard.c:38): No such file or directory
[E] os_resetusb (via usb.c:169): resetusb failed: No such device
[E] usb_tryreset (usb.c:177): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[1]    14894 killed     sudo ckb-daemon
```

Thanks.

sirjaren, 2016-04-25:
More information (K70 RGB):

<pre>
<b>$ dmesg</b>
41155.611614] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.0/0003:1B1C:1B13.001E/input/input53
[41155.662746] hid-generic 0003:1B1C:1B13.001E: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:00:1d.0-1.4/input0
[41165.663976] hid-generic 0003:1B1C:1B13.001F: usb_submit_urb(ctrl) failed: -1
[41165.663996] hid-generic 0003:1B1C:1B13.001F: timeout initializing reports
[41165.664289] input: Corsair Corsair K70 RGB Gaming Keyboard  as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.1/0003:1B1C:1B13.001F/input/input54
[41165.714859] hid-generic 0003:1B1C:1B13.001F: input,hiddev0,hidraw4: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard ] on usb-0000:00:1d.0-1.4/input1

<b>$ lsusb -v -d 1b1c:1b13</b>
Bus 002 Device 010: ID 1b1c:1b13 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b13 
  bcdDevice            2.04
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     112
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
</pre>

ccMSC, 2016-04-26:
Whoops...for some reason I had set the endpoints to be device-specific instead of firmware-specific. Should be fixed now.

The commit I just pushed will log *a lot* of lines, so if it's not working, just send me everything up until the "attempting reset" part.

jbickers, 2016-04-26:
The new build proceeds further, which is great, but once ckb starts then keyboard input is disabled. So ckb-daemon starts ok, with no visible change to the state of the keyboard, then when ckb starts my colour profile gets set correctly, but keyboard input is gone. Closing ckb restores keyboard input.

I've tried without usb.quirks, with the normal K70 RGB usb.quirks, and with usb.quirks=0x4. The results were the same. The log output below is from the usb.quirks=0x4 test.

The first part is the setup, before ckb is started.

```
    ckb: Corsair RGB driver beta-v0.2.5+f03
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[I] Connecting 3 endpoints
[I] Claiming EP 0
[I] Claiming EP 1
[I] Claiming EP 2
[I] Starting input thread for /dev/input/ckb1
[I] Sending URB_CONTROL to ep 2
[I] Receiving URB_CONTROL from ep 2
[E] os_usbrecv (via firmware.c:15): Connection timed out
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
```

This fragment is from syslog when the daemon starts.

```
Apr 26 22:54:49 jb2016 acpid: input device has been disconnected, fd 20
Apr 26 22:54:50 jb2016 acpid: input device has been disconnected, fd 21
Apr 26 22:54:50 jb2016 kernel: [ 1041.013773] input: ckb1: Corsair K70 RGB Gaming Keyboard as /devices/virtual/input/input46
Apr 26 22:54:50 jb2016 kernel: [ 1041.014102] input: ckb1: Corsair K70 RGB Gaming Keyboard as /devices/virtual/input/input47
Apr 26 22:54:55 jb2016 kernel: [ 1046.153902] usb 4-2: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
```

This next bit is the daemon log when ckb is started. It's not endless; there's a burst of these, then no more.

```
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
```

And finally, when ckb is stopped, there's another burst of the URB_INTERRUPT messages.

```
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
[I] Sending URB_INTERRUPT to ep 3
```

sirjaren, 2016-04-27:
Confirming again what @jbickers has.  We're almost there. :smile:

tatokis, 2016-04-27:
@ccMSC Since you mentioned you do not have a K70 RGB, I thought it would be useful to you if I sniffed the usb packets.

Sniffing was performed with firmware versions 1.30 and 2.04, and in both cases the keyboard was already plugged in, while capturing started right before running ckb-daemon.

Since I noticed that while animations are now working, ckb-daemon no longer recognises keystrokes, I also captured the data sent when pressing "1234567890" on the keyboard.

https://vps.tasossah.com/bugs/ckb_wireshark.tar.gz

ccMSC, 2016-04-28:
Thanks! I removed the logging because it looks like the setup phase is working now. Updated the input code based on @tatokis 's USB captures - hopefully it works this time.

egar3427, 2016-04-28:
work fine, thanks

4t0m1k, 2016-04-28:
Great ! How long for the 0.26 ckb mac os x binary to be released ? If it's too long, I'll try to build it, but I prefer to get the binary directly :) Thanks !

tatokis, 2016-04-28:
I just pulled in the new commits and I can confirm that input is working again.
One thing I noticed from the daemon log is this:
```
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
```
I tried saving to the hardware profile and everything went as planned, so I don't think any functionality is affected by the above.

tatokis, 2016-04-28:
@ccMSC I found an issue, setting the keyboard in BIOS mode causes ckb-daemon to segfault.
```
Starting program: /usr/bin/ckb-daemon 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
    ckb: Corsair RGB driver beta-v0.2.5+f04
[I] Root controller ready at /dev/input/ckb0
[New Thread 0x7ffff75e8700 (LWP 12825)]
[I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
[New Thread 0x7ffff6de7700 (LWP 12838)]
[I] Starting input thread for /dev/input/ckb1
[New Thread 0x7ffff65e6700 (LWP 12839)]

Thread 3 "ckb-daemon" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff6de7700 (LWP 12838)]
0x00000000004093eb in os_inputmain (context=0x8) at usb_linux.c:105
105	        urbs[2].buffer_length = MSG_SIZE;
(gdb) bt full
#0  0x00000000004093eb in os_inputmain (context=0x8) at usb_linux.c:105
        kb = 0x623998 <keyboard+248>
        fd = 3
        vendor = 6940
        product = 6931
        index = 1
        urbcount = 0
        urbs = 0x7fff00000015
#1  0x00007ffff7bc16fa in start_thread (arg=0x7ffff6de7700)
    at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7ffff6de7700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737335162624, 
                4151005133144482773, 0, 140737343553103, 140737335163328, 0, 
                -4150990027181628459, -4150988082954607659}, 
              mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, 
            data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
---Type <return> to continue, or q <return> to quit---
#2  0x00007ffff78f7b5d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
No locals.
(gdb) quit
```
And here's a capture from the keyboard initialising with ckb-daemon in bios mode.
https://vps.tasossah.com/bugs/ckb_204_init_BIOS.pcapng.tar.gz

My guess is that ckb-daemon gets a packet bigger than what it expects, and segfaults.

phorgan1, 2016-04-28:
I just tried the latest testing as of now and no joy. I found that killing the ckb does not make the keyboard start working but killing the daemon lets it work immediately.

tatokis, 2016-04-29:
Do you mean the testing branch? 
The fixes are in the fw2kb branch.

jbickers, 2016-04-29:
Good work, the fix works for me.

duythanh90, 2016-04-29:
After update my K65 RGB to fw 2.04 CKB does not work anymore :( 
CKB Application still detect my device but I can not type or change the keys color.
But if I quit the CKB application I can still use my K65 as a normal keyboard with normal settings.

I don't know much about the Terminal things :( 
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.5
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 RGB Gaming Keyboard  at /var/run/ckb1
[I] Adding ckb1.ifusb[2]
[I] Adding ckb1.ifhid[1]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[I] Adding ckb1.ifhid[0]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Reading ckb1.ifhid[0]
[I] Reading ckb1.ifhid[1]
[E] loadrgb_kb (led_keyboard.c:211): Bad input header
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Setup finished for /var/run/ckb1

4t0m1k, 2016-04-29:
So, I managed to build ckb-fw2kb and it works perfectly ! Didn't have to download any third library, only XCode and Qt5 (that I already have both), this is great !

Thanks !

LiamMayfair, 2016-04-30:
Confirmed Corsair K70 RGB v2.04 working perfectly again on Fedora 23 kernel 4.4.8 after installing fw2kb branch. Thanks a lot!

mhennessie, 2016-05-01:
@4t0m1k can you post your build for OS X? I have a K65 RGB that I downgraded to 1.33 because 2.04 stopped it from working at all when CKB was running. Would love to have it updated again.

4t0m1k, 2016-05-01:
@mhennessie

[ckb.app.zip](https://github.com/ccMSC/ckb/files/244221/ckb.app.zip)

Copy ckb.app in you /Applications folder, then reboot.

mhennessie, 2016-05-01:
@4t0m1k 

Thanks but it won't launch. Maybe I will give building it myself a try since I have all the tools installed anyways.

4t0m1k, 2016-05-01:
Maybe you have to do this :

Copy `ckb.app` to your Applications folder. Copy the file  [`service/launchd/com.ckb.daemon.plist`](https://raw.githubusercontent.com/ccMSC/ckb/master/service/launchd/com.ckb.daemon.plist) to your computer's `/Library/LaunchDaemons` folder (you can get to it by pressing `Cmd+Shift+G` in Finder and typing the location). Then open a Terminal and run the following commands to launch the driver:

```
sudo chown root:wheel /Library/LaunchDaemons/com.ckb.daemon.plist
sudo chmod 0700 /Library/LaunchDaemons/com.ckb.daemon.plist
sudo launchctl load /Library/LaunchDaemons/com.ckb.daemon.plist

-----------------

Building by yourself is very easy if you have Xcode and Qt5, in Terminal, go to the ckb-fw2kb folder (from .zip that you download on github) and launch `./quickinstall`

If your Qt installation is not the default one, you have to declare the QMAKE env var. My command for example : `QMAKE=/Developer/Qt/5.5/clang_64/bin/qmake ./quickinstall`

Then, at the end of the install, it will propose you to install it, press 'y' 3 times.

mhennessie, 2016-05-01:
Thanks. I'll try fixing the permissions and if that doesn't work I'll try to build it.

mhennessie, 2016-05-02:
@4t0m1k - Thanks for your help on this. I ended up building it myself and it works fine with my K65 RGB running 2.04. Installing it this way also fixed an issue I have had using CKB for a while, the menubar icon has been blank since upgrading to El Capitan last fall. The menubar icon has now been restored, yay.

mhennessie, 2016-05-02:
For anyone that needs it I have posted a copy of the 1.33 firmware for the K65 RGB here: http://forum.corsair.com/v3/showpost.php?p=848745&postcount=136

I got it from Corsair support when I submitted a ticket for 2.04 not working in Windows. That thread also has the 1.30 firmware for the K65/70/95 RGB.

spacekookie, 2016-05-02:
I reinstalled my computer yesterday and after installing ckb it no longer recognises my keyboard, even though it's plugged in and works.
The troubeling thing is that something overwrites my OSes keyboard layout (de) to en_US -.-

I'm running Linux Manjaro 15, with CKB version 0.2.5.
If you need more info to help you fix this, I'll happily provide it.

Mills00013, 2016-05-02:
Anyone able to get the keyboard working under Arch with the fixes committed in the fw2kb branch? Compiled successfully here but still doesn't seem to pick up the keyboard.

`
ckb-daemon[3151]:     ckb: Corsair RGB driver beta-v0.2.5+f04
ckb-daemon[3151]: [I] Root controller ready at /dev/input/ckb0
ckb-daemon[3151]: [I] Connecting  at /dev/input/ckb1
ckb-daemon[3151]: [W] os_setupusb (usb_linux.c:293): Unable to read endpoint count from udev, assuming 4...
ckb-daemon[3151]: [E] os_setupusb (usb_linux.c:296): Failed to claim interfaces: No such file or directory
ckb-daemon[3151]: [I] Disconnecting /dev/input/ckb1
ckb-daemon[3151]: [I] Removed device path /dev/input/ckb1
ckb-daemon[3151]: [I] Connecting Corsair M65 RGB Gaming Mouse at /dev/input/ckb2
ckb-daemon[3151]: [I] Starting input thread for /dev/input/ckb2
ckb-daemon[3151]: [I] Setup finished for /dev/input/ckb2
`

BradAbard, 2016-05-02:
I compiled the latest fw2k branch and got this:
```
iMac:~ INGLIS$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ckb: Corsair RGB driver beta-v0.2.5+f04
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb1
[I] Connecting Corsair K70 RGB Gaming Keyboard  at /var/run/ckb2
[I] Adding ckb2.ifhid[0]
[E] iterate_devices_hid (usb_mac.c:731): Failed to create device plugin: e00002be
```
Still the same error as #233 :(

Is there an easy way to turn logging on?

4t0m1k, 2016-05-03:
@BradAbard Did you plug the power usb cable first, then the keyboard usb cable ? And have you tried to only plug the keyboard usb cable ? I can't get my keyboard to work if I plug both the cables, but only with the keyboard usb cable, it works very well. I know you are on USB2, but give it a try.

duythanh90, 2016-05-05:
Update: Thanks @mhennessie  for upload the FW 1.33
- I Downgrade my K65 to FW 1.33 and using CKB version 0.2.5 everything seem fine instead of the led effect look a bit laggy 
So I Uninstall CKB and re-install version 0.2.4 and everything is work fine now :)

fliiiix, 2016-05-05:
Just fyi: on fedora 22 (4.4.6-201.fc22.x86_64) the K70 RGB works fine with the origin/fw2kb branch :+1:

hxltinh, 2016-05-08:
I'm too, using K70 RGB with OSX 10.11.3, I've bought a keyboard yesterday and I could not make the ckb recognized my keyboard. I've already update firmware to latest with window PC.

[25 more replies]

closed 2016-06-13 by ccMSC