ccMSC/ckb #301: Less than 33%

JonoB, 2016-04-21:
Hey guys,

I've just installed on ubuntu, and its working perfectly. Quick question:there are 4 brightness modes: 0%, 33%, 67% and 100%

Is it possible to create another brightness percentage, like 15% or 20%?

ccMSC, 2016-04-24:
Not currently. You could set all the keys to black and then decrease the `Opacity` field on the animation, that's the closest I can think of.

JonoB, 2016-04-24:
Is this a limitation of the keyboard (I have a Strafe) or a limitation of the software?

EvN-SKO, 2016-04-24:
@JonoB @ccMSC i believe this is a hardware limitation, ask corsair for confirmation

ccMSC, 2016-04-26:
No, although the older keyboards (K65, K70, K95) have a very limited color range so adding more steps wouldn't make much sense. I've considered adding this feature at some point but it's not going to be a priority.

fleischie, 2016-09-06:
Old thread maybe, but you could manually set the dimming via the `ckb-daemon`.

I elaborated on this in [Issue 451](https://github.com/ccMSC/ckb/issues/451#issuecomment-244678928) but I must confess, that this might be a little more work for simpler setups.

It is possible though. 😉