ccMSC/ckb #302: K65 RGB not being recognized

AustinSwagner, 2016-04-22:
While in bootcamp the other day I updated the CUE to the latest version and everything went smoothly in bootcamp. Yet, booting into OS X the keyboard about half the time will get recognized by CKB but not show any color presets and the keys are not remapped(Alt to Command). The keyboard works fine in bootcamp but is only is red with the wasd and the arrow keys white in OS X and only inputs text sometimes. I have the keyboard firmware updated to 2.04 and CKB is beta 0.2.5. Please help I want my keyboard to work. Thank You!

plan8studios, 2016-04-23:
I had the same problem with 0.2.5 and I had to downgrade to 0.2.4 (https://github.com/ccMSC/ckb/releases/tag/v0.2.4). It's still buggy, but at least it sees the keyboard and remembers my presets... Most of the time.

AustinSwagner, 2016-04-23:
With that version the keyboard does not show up at all but in working just fine. Still red with white wasd and arrows.

plan8studios, 2016-04-23:
Mine doesn't show up right now either. Like I said, it's buggy :-)

Sometimes the animations just stop working. Sometimes flicking the switch on the back will help. Other times - like tonight - doing that causes the keyboard to become completely unresponsive and it must be un/re-plugged to get it working. And sometimes, killing the ckb-daemon process helps. Other times, the computer goes into a death spiral of endless keypresses. And, for whatever reason, after I kill the daemon and it restarts, it seems to use an awful lot of CPU -- around 50% on a 2.7GHz i7. The best bet I've found for "fixing" it is a reboot.

I'm not sure I'm at the newest firmware on the keyboard, which could be part of the issues I'm having, or maybe why I can sometimes get it to work right ;-) It's times like this I wish I knew C++ and could dig into the source and figure out what the fuck is going on.

AustinSwagner, 2016-04-23:
Yeah for real I don't know anything about coding. I did update the firmware of the keyboard in bootcamp to 2.04. If you can get the keyboard to show up in ckb it will tell you under the device tab, I think. I had this problem even when on the older keyboard firmware.

plan8studios, 2016-04-23:
I just upgraded the firmware... And now it's in that hideous red and white and it's still not showing up in ckb. Even rebooted, nothing. ugh. I guess I'll try 0.25 again and hope it doesn't thrash my CPU too hard.

AustinSwagner, 2016-04-23:
Good luck to you! In the meantime I'm using my good old apple wireless keyboard. Hooray!

plan8studios, 2016-04-23:
Thanks :) I wish I could take this damn thing back to BestBuy but it's been too long. So I'm kind of stuck with it. I doubt the wife would let me drop *another* $100+ on *another* keyboard.

ccMSC, 2016-04-24:
The latest firmware isn't compatible with ckb - see #298.

plan8studios, 2016-04-24:
I noticed that after I upgraded :( Anything I could do to help? I grabbed the source and installed QT creator last night. However, after about an hour of poking around the code and realizing I know nothing about C++, QT, or USB programming, I gave up.

mhennessie, 2016-05-03:
You just need to grab the source for the fw2kb branch and build that. It works fine in OS X with the 2.04 firmware. If you don't want to do that then go to the Corsair forums and you can find firmware 1.30 and 1.33 for the K65 RGB. Using CUE you can downgrade it.

plan8studios, 2016-05-03:
I recently upgraded to 2.04 (or whatever the newest is from CUE) and built the fw2kb branch, and my keyboard still isn't recognized :( It's usable, it works, and my kep remappings are recognized, but the GUI doesn't see the keyboard.

closed 2016-06-13 by ccMSC