ccMSC/ckb #304: Ubuntu 16.04 LTS and Manjaro Linux / Archlinux: Keyboard became non-responsive after starting the ckb software

stnight, 2016-04-23:
Please help me. I installed the ckb then my keyboards stops working. Is there any work around for this issue? I've already change my polling rate, but whenever I did, the ckb can't now detect my keyboard.

Gabscap, 2016-04-23:
Same error. I switched to Windows and updated the firmware and switched back to Linux and as soon as the ckb daemon starts, the keyboard does not work.

OS: Ubuntu 15.10/16.04
Corsair K70 RGB
[Syslog](http://pastebin.com/raw/VHXZjLN8) (there were multiple restarts)
ckb version: 33734b2

I disabled the ckb-daemon as a workaround.

stnight, 2016-04-23:
I disabled it too, but the point is I can't create customized light effects and stuff :/

ccMSC, 2016-04-24:
The current firmware isn't compatible with ckb. See #298 (someone also posted link for older firmware in that thread, so that might be able to help you).

closed 2016-04-24 by ccMSC

stnight, 2016-04-24:
@ccMSC so we're gonna wait for the next build? :) thanks for referring me to the other same issue.