ccMSC/ckb #305: K70 RGB not responding to key presses on OS X

lhecker, 2016-04-23:
Last week ago ckb still worked flawless on OS X (for the first time for me even!). I believe after the new Firmware came out (2.04) it's bugged though.

Let's say I (force-)kill the ckb-daemon service... It doesn't matter whether ckb itself is running or not: Key presses only work for the first couple seconds! I noticed that the keyboard stops responding as soon as ckb shows the detected keyboard (using the graphical representation).
If however I switch the polling rate on the keyboard's back side right after the ckb-daemon starts up it works completely again without breaking. Interestingly the ckb app show's the keyboard as undetected then.

Tell me if you need to know any specs from my PC, the keyboard, or if you need to run any tests. :blush:

ccMSC, 2016-04-24:
See #298. I'll hopefully have an updated testing version in the next day or two.

closed 2016-04-24 by ccMSC

lhecker, 2016-04-24:
Oh thanks, @ccMSC! I'm very sorry for having created this unnecessary issue then. :sweat_smile: