ccMSC/ckb #308: Ubuntu 16.04 crash

vooze, 2016-04-24:
Hi.

First: Thank you so much for this, I even donated to show how much your work means to me!

But after switching to ubuntu 16.04 (fresh install)  ckb often crashes.

[image]

Do you know whats going on? Are there some logs I can show you?

ccMSC, 2016-04-26:
Somebody else reported this as well - could you provide a gdb backtrace?

Compile ckb with `./qmake-auto && make debug` and then run with `gdb bin/ckb`. When the program crashes, enter `bt full` and paste the output here.

vooze, 2016-04-26:
Hi @ccMSC.

I found https://github.com/ccMSC/ckb/issues/296 after posting, feel a bit stupid to not search first. Anyway I did like @tatokis said. No crashes so far for 24 hours.

ccMSC, 2016-04-28:
Okay, thanks. If it happens again please post the backtrace in #296 :)

closed 2016-04-28 by ccMSC