ccMSC/ckb #310: Windows Key Mapping not working

RobErskine, 2016-04-24:
I'm attempting to map the Windows key (in between alt and control) to the Left Option key on Mac OSX 10.10.3, but it doesn't seem to be registering using ckb beta-v0.2.5. The key illuminates on keypress but isn't registering in any applications. 

I have this setup in:

- my [Mac OSX Keyboard config](http://cl.ly/2a1M1U1o0L23/Screen%20Shot%202016-04-24%20at%206.36.54%20PM.jpg)
- [ckb's global settings](http://cl.ly/0j0r1J3y2f0Y/Screen%20Shot%202016-04-24%20at%206.37.49%20PM.jpg)
- and in my [particular profile binding setting](http://cl.ly/1C2P1M0N3A43/Screen%20Shot%202016-04-24%20at%206.38.09%20PM.jpg)

The key used to work fine, and still does when I use my laptops keyboard, just not with ckb. Am I missing something super silly?

plan8studios, 2016-04-25:
I had problems with this at first, too. The only thing that worked for me was setting up the binding, like you have in your 3rd screenshot. I can't remember exactly what I did to get it working, but it is. Have you tried also remapping the alt key to the left cmd key? Or clicking the "Save to hardware" button? Or un/re-plugging the keyboard?

RobErskine, 2016-04-25:
@plan8studios thanks for those tips. My left alt is bound to left cmd and it works just fine. I tried your other options too, but I didn't get anywhere. I kept digging and found that when I moved it over to a new profile (from rainbow -> breathing) and on breathing it works fine. I guess I can just nuke rainbow and use this for now. 

Let me know if there is a way to export profiles, I'd love to help debug

ccMSC, 2016-04-26:
Sounds like it's an issue with the default binding. The modifier mappings are due for an overhaul anyway - I'll see if I can find the problem.

RasmusVoss, 2016-06-15:
I see it to ckb-betav0.2.5 out of nowhere the binding stopped working yesterday. 
Is there anything we add to the debugging of this issue ?

wwvuillemot, 2016-08-16:
I had a similar issue with the Windows keep (mapped to Option) stopped working.  Note, this is on Mac OS X 10.11.6

Resolution.

1. Duplicate the affected Mapping
1. Rename the old Mapping (if you do not do this, at least on Mac OS X v.26 it will crash when you try to delete)
1. Delete the old mapping

Basically, just duplicating the mapping seemed to unstick it.  I tried everything else from re-installing CKB, to unplugging keyboard, restarting Mac OS X, to binding different keys ... nothing worked but the above.  And the issue started rather suddenly.  The only thing I recall different lately is Logitech auto-updated its drivers for my G502 ... coincident or causal?  I do not know.

closed 2018-03-30 by RobErskine